pabr/leansdr #2: GUI

ad6sh, 2017-03-19:
could you please tell how to run the gui of leandvb

pabr, 2017-03-19:
Compile on a host with X11 development libraries; start with --gui.
Note: The gui is a debugging tool, not a replacement for command-line options.

closed 2017-03-19 by pabr

ad6sh, 2017-03-20:
Than you so much!. We have seen the sinal transmitted in spectrum analyzer..but the bandwidth is about 2Mhz.The signal constellation contains large amount of noise also.could you please help to solve the problem??

ad6sh, 2017-03-20:
SORRY,THE BANDWIDTH IS ABOUT 2OMhz

pabr, 2017-03-20:
Sample at 24 MHz or more.
Configure --f, --sr, --cr to match the parameters of the signal and the settings of the receiver.
Run "leandvb -v" and check in the output that the parameters are OK.
Record to a file and run leandvb offline.
If you see a QPSK constellation but too much noise, add "--sampler rrc --viterbi".

ad6sh, 2017-03-20:
Symbol rate 125ksymbols/sec ,fec 7/8,pid 255,gain 5 ,i've set like this transmitter side and got 8MHz , i've mailed the photos,please do go through it