pabr/leansdr #3: Not getting QPSK spectrum

ad6sh, 2017-03-22:
[image]
Bonjour, we are getting a spectrum as attached while transmitting signals with following particulars
sr-125ks/s,,f-98Mhz,,fec-1/2,,gain 7,, output mode QPSKRF, We are transmitting jpeg pictures using rpidatv

pabr, 2017-03-22:
The top-left "rawiq" plot shows that your signal is heavily saturated.
Please read http://www.pabr.org/radio/leandvb/leandvb.en.html#ts

ad6sh, 2017-03-22:
Merci beacoup !

ad6sh, 2017-03-22:
we are still having the same problem even though we tried different gains.

F5OEO, 2018-01-28:
Seems there is no modulation. Rpidatv doesn"t use jpeg input but only transport stream. Could be the problem.

ad6sh, 2018-01-29:
Sui, Merci Beaucoup :)