pabr/leansdr #9: OSX Build failure

cpbridges, 2017-09-08:
I've tried adding -I/opt/X11/include, as well as exporting to $PATH and env CPPFLAG options but these don't work. Is there any chance someone could give me some more options to edit the makefile for OSX/mac?

Many thanks in advance, Chris

```
Chris-MacBookAir:apps chrisbridges$ make
g++ -O3 -I.. -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DGUI leandvb.cc -lX11 -o leandvb  ||  \
	g++ -O3 -I.. -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable       leandvb.cc       -o leandvb
In file included from leandvb.cc:17:
../leansdr/gui.h:16:10: fatal error: 'X11/X.h' file not found
#include <X11/X.h>
         ^
1 error generated.
In file included from leandvb.cc:14:
../leansdr/sdr.h:61:26: error: variable length array of non-POD element type
      'complex<float>'
      complex<float> data[fft.n];
                         ^
../leansdr/sdr.h:52:4: note: in instantiation of member function
      'leansdr::auto_notch<float>::detect' requested here
          detect();
          ^
../leansdr/sdr.h:33:5: note: in instantiation of member function
      'leansdr::auto_notch<float>::run' requested here
    auto_notch(scheduler *sch, pipebuf< complex<T> > &_in,
    ^
leandvb.cc:265:11: note: in instantiation of member function
      'leansdr::auto_notch<float>::auto_notch' requested here
      new auto_notch<f32>(&sch, *p_preprocessed, *p_autonotched,
          ^
In file included from leandvb.cc:14:
../leansdr/sdr.h:1198:22: error: variable length array of non-POD element type
      'complex<float>'
      complex<T> data[fft.n];
                     ^
../leansdr/sdr.h:1187:4: note: in instantiation of member function
      'leansdr::cnr_fft<float>::do_cnr' requested here
          do_cnr();
          ^
../leansdr/sdr.h:1166:5: note: in instantiation of member function
      'leansdr::cnr_fft<float>::run' requested here
    cnr_fft(scheduler *sch, pipebuf< complex<T> > &_in, pipebuf<float> &_out,
    ^
leandvb.cc:292:17: note: in instantiation of member function
      'leansdr::cnr_fft<float>::cnr_fft' requested here
    r_cnr = new cnr_fft<f32>(&sch, *p_preprocessed, p_cnr, cfg.Fm/cfg.Fs);
                ^
2 errors generated.
```

cpbridges, 2017-09-08:
Got a little further. Added to Makefile:

1. CXXFLAGS line has -I/usr/X11R6/include
2. And -DGUI line has -lX11 -L/usr/X11R6/lib.

Only errors are now "error: variable length array of non-POD element type"

Best, Chris

cpbridges, 2017-09-08:
Solved. Found the error is compiler based with clang. Use homebrew's gcc4.9 and appears to work now. Replace g++ with /usr/local/Cellar/gcc/4.9.2_1/bin/g++-4.9 in the Makefile to complete. Best, Chris

closed 2017-09-08 by cpbridges

cn0xroot, 2017-09-20:
@cpbridges hi chris, 
i had meet the same problem
can you Paste the code of Makefile after you modify？thanks！

cn0xroot, 2017-09-20:
fix x11 path 

```
cat Makefile
APPS	= leandvb leansdrscan
APPS	+= leansdrcat leantsgen leanchansim leandvbtx

all:	$(APPS)

clean:
	rm -f $(APPS)

DEPS	= ../leansdr/*.h

CXXFLAGS	= -O3 -I.. -I/opt/X11/include  -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable

%:	%.cc $(DEPS)
	g++ $(CXXFLAGS) -DGUI $< -lX11 -L/opt/X11/lib -o $@  ||  \
	g++ $(CXXFLAGS)       $<       -o $@

EMBED_FLAGS=	-I.. -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable  \
		-Ofast -mfpu=neon -funsafe-math-optimizations -fsingle-precision-constant

leandvb.embedded:	leandvb.cc $(DEPS)
	g++ $(CXXFLAGS) $(EMBED_FLAGS) $< -static -o $@  ||  \
	g++ $(CXXFLAGS) $(EMBED_FLAGS) $<         -o $@
```
Solved.