pabr/leansdr #10: Autonotch Loop

cpbridges, 2017-09-17:
Hi Pabr,

I'm Tx'ing using a USRP QPSK 1/2 rate signal at 1M symbol rate - and Rx'ing using a TXCO RTL dongle at 2M sample rate. Followed all advise on your pair pages but I'm in a forever loop in the auto notch filter - any ideas? 73s Chris

Commands:
```
rtl_sdr -g 40 -f 1.28G -s 2000000 rec.fifo
./leandvb --gui -v -d -f 2000e3 --sr 1000e3 --cr 1/2 --derotate -5000 --anf 2 < ~/rec.fifo > - | mplayer -cache 128 -
```
Output:
```
auto_notch: slot 0 new peak 4003 -> 101
auto_notch: slot 1 new peak 3962 -> 3713
auto_notch: slot 0 new peak 101 -> 129
auto_notch: slot 1 new peak 3713 -> 206
auto_notch: slot 0 new peak 129 -> 3696
auto_notch: slot 1 new peak 206 -> 3718
auto_notch: slot 0 new peak 3696 -> 3683
auto_notch: slot 1 new peak 3718 -> 3730
auto_notch: slot 0 new peak 3683 -> 3813
auto_notch: slot 1 new peak 3730 -> 3416

```
[image]

pabr, 2017-09-17:
Your signal looks clean, the spectrum has a reasonably flat top; that's why the ANF wanders around, and this should be harmless. Have you tried "--anf 0" ?

ANF will probably be disabled by default in a future release.

cpbridges, 2017-09-17:
Thanks for the fast response! It's super flat as I'm going straight in via a short sma cable (just for test).

I've tried with "--anf 0" and the wandering goes away but no video yet.

I've put up my flow graph too. I'm sure some others would find it useful. The .ts file is the one you recommend too.

In the gui - I couldn't explicitly find out what the lines each meant. Blue is obvious, but SS and MER? And the yellow? Interesting criss-crossing of the red... unsure what that means.

Chris

[image]
[image]

pabr, 2017-09-17:
SS is signal strength; I have no idea why it is wobbling like that.
MER is Modulation Error Ratio, a kind of SNR for constellation-based modulations.
Yellow is an estimation of the percentage of recovered packets.

Your flowgraph seems to be a modulator for DVB-S2. Unfortunately leandvb cannot decode S2 yet. Hopefully I will implement it before Es'hail2 launches.

cpbridges, 2017-09-17:
Ah! That'd be it then! I'll try DVBS then and see how we go. Will report back in a little bit. Chris

cpbridges, 2017-09-17:
Almost there, modded gr-dvbs tx grc file to Tx properly. See attached. Almost there - will try some mplayer settings next. Suggestions would be great still though!
[image]
[image]
[image]
[image]

cpbridges, 2017-09-17:
Here's where it's at, comments welcome on the last hurdles. Will close tomorrow. Best, Chris

[https://www.youtube.com/watch?v=JPkJTpsf5CM](url)

pabr, 2017-09-17:
Yes, recovery is incorrectly reported above 100% for some combinations of settings.

In your video there are sequences where leandvb fails to lock despite a very good MER (tight constellation clouds). This should not happen. You may want to:
- Check whether the CPU is overloaded (note: leandvb is single-threaded)
- Try the "--fastlock" option (increases CPU usage)
- Record IQ samples to a file, then demodulate afterward
- Or pipe directly from rtl_sdr to leandvb (when writing to a file, rtl_sdr may silently lose samples if the OS pauses while flushing its cache).

There are other sequences with poor MER (wide constellation clouds). To demodulate that you need the "--viterbi" option (CPU-intensive).

Regarding mplayer, I find that it is best to specify as many parameters as possible rather than rely on auto-detection, as in:
mplayer  -demuxer mpegts  -tsprobe 1  -vid 256  -aid 257  -vc ffmpeg2  -ao null  -
To check whether the TS is good, record it to a file and run mplayer afterward.
Maybe the decoder that F5OEO selected for rpidatv (omxplayer ?) is a better choice for live streams.

cpbridges, 2017-09-18:
Understood - many thanks par. 73 C

closed 2017-09-18 by cpbridges