pabr/leansdr #11: Build failure on openSUSE Leap 4.3

tpbaker, 2018-01-28:
I get errors when attempting to compile , , odd.

tim@linux-8m91:~/src/leansdr/src/apps> make
g++ -O3 -I.. -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DGUI leandvb.cc -lX11 -o leandvb  ||  \
g++ -O3 -I.. -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable       leandvb.cc       -o leandvb
/tmp/ccaWZX5A.o:(.rodata._ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE[_ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE]+0x10): undefined reference to `leansdr::viterbi_dec_interface<unsigned char, unsigned char, int, int>::update(int*, int*)'
/tmp/ccaWZX5A.o:(.rodata._ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE[_ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE]+0x18): undefined reference to `leansdr::viterbi_dec_interface<unsigned char, unsigned char, int, int>::update(unsigned char, int, int*)'
collect2: error: ld returned 1 exit status
/tmp/ccU7EsXK.o:(.rodata._ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE[_ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE]+0x10): undefined reference to `leansdr::viterbi_dec_interface<unsigned char, unsigned char, int, int>::update(int*, int*)'
/tmp/ccU7EsXK.o:(.rodata._ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE[_ZTVN7leansdr21viterbi_dec_interfaceIhhiiEE]+0x18): undefined reference to `leansdr::viterbi_dec_interface<unsigned char, unsigned char, int, int>::update(unsigned char, int, int*)'
collect2: error: ld returned 1 exit status
Makefile:14: recipe for target 'leandvb' failed
make: *** [leandvb] Error 1
tim@linux-8m91:~/src/leansdr/src/apps> gcc --version
gcc (SUSE Linux) 4.8.5

pabr, 2018-01-28:
Should be fixed now (tested with gcc 4.6.3), please confirm.

tpbaker, 2018-01-28:
Confirmed fixed. Just compiled with zero errors. Thanks much !

closed 2018-01-28 by tpbaker