pabr/leansdr #13: leaniiorx: unknown error

jh4xsy, 2018-09-01:
Hi

I ran leaniiorx, then got following error.

    % ./leaniiorx -f 2395e6 --bw 2000e3 -v
    Setting RF bandwidth 2000 kHz
    ** IIO: in_voltage_sampling_frequency failed with Unknown error -2

I'm using PlutoSDR v0.27 with libiio 0.15 on openSUSE leap 15 x86-64 machine.

    % iio_info -s
    Library version: 0.15 (git tag: 6ecff5d)
    Compiled with backends: local xml ip usb serial
    Available contexts:
            0: Local devices [local:]
            1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104400b83991000910001f00040c6539e3 [usb:3.8.5]

pabr, 2018-09-01:
The "%" prompt suggests you running leaniiorx on the x86 host. This might work thanks to the IIO/USB abstraction, but I have never tested it.

Did you run another IIO application before leaniiorx ? Maybe the SDR is in a state that leaniiorx fails to handle properly.

Does running "dmesg" on the PlutoSDR reveal details about the error ?

Do you get the same error when running leaniiorx natively on the PlutoSDR ?

pabr, 2018-09-01:
Note that you will need to specify --nbufs to take advantage of the 230 MB and also to work around a bug (missing default value).

Keeping this issue open until I test over IIO/USB.