pabr/leansdr #14: DVB-S2 rx - not getting .ts output

cgavrila1, 2019-02-16:
Hi,

I am trying to use the leandvb as an offline DVB-S2 receiver using an I/Q file as input.
The file was generated using the https://github.com/drmpeg/gr-dvbs transmitter. 

The command used to run leandvb is the following:
`./leandvb --f32 -f 2e6 --sr 500e3 --cr 9/10 --gui --fd-info 2 --anf 1 --standard DVB-S2 --const 16APSK  --sampler rrc < recorded_data_source.iq > capture.ts`

But I get no output written into capture.ts. Here are some screenshots:
[image]

The preprocessed I/Q looks fine but the constellation does not.
I can not upload the recorded_data_source.iq as it is has a large size (~1.9GB).

The overall idea would be to use 2 USRPs (tx <-> rx) to transmit a .ts file. I have the chain available but when I redirect the USRP's source to a file then pass it to leandvb it gets even worse:
[image]

Any help would be highly appreciated. Please let me know if further info in needed from my side. Thank you very much

pabr, 2019-02-16:
Are you sure about "-f 2e6 --sr 500e3" ? In the spectrum view the signal seems to occupy half the bandwidth, so maybe try "--sr 1e6".

Also, for DVB-S2, you don't need to specify --cr nor --const.

And you mention gr-dvbs, but I suppose you mean gr- dvbs2

cgavrila1, 2019-02-16:
Oh god, terribly sorry, got mixed up with the input files and lean dvb samples. Adapted to "-- sr 1e6" all good now, thank you very much.

P.S. Indeed I meant gr-dvbs2, another mistake from my side.

Thanks a lot for you help.

closed 2019-02-16 by cgavrila1