pabr/leansdr #15: How to provide live input from USRP

cgavrila1, 2019-02-19:
Hi again,

I am trying to provide a live input from an USRP device to leandvb input. Basically I am trying to achieve something like:
`rtl_sdr  ... -  |   ./leandvb  --gui  ...`

For starters I was thinking to redirect the USRP's outptu to an udp port and then, using `nc -u ..` to provide this as input for leandvb, sadly with no success.

Wanted to ask you if it is possible for leandvb to get the input data from a local udp port or, if I'm way off the track, if you may have an advice on how can I feed live data from USRP to leandvb.

I read on http://www.pabr.org/radio/leandvb/leandvb.en.html that: " Instead, third-party applications can spawn a leandvb process, inject I/Q samples into its standard input, ..." but I am not sure how to achieve this.

Thank you very much

pabr, 2019-02-19:
leandvb only reads IQ from standard input, in uint8, int16 or float32 complex format. You should ask the manufacturer of your device whether they provide a command-line tool similar to "rtl_sdr".

Note that UDP is a poor choice for this application. Analog modulations may tolerate lost packets, but not DVB. UDP will look like it works in practice, until it doesn't, and then you'll waste hours diagnosing the problem.

cgavrila1, 2019-03-04:
Thank you very much for your help.

closed 2019-03-04 by cgavrila1