pabr/leansdr #16: output bb frame

cgavrila1, 2019-03-16:
Hi there,

Just a small question, is it possible to output the bb frames directly? What I want to do is to provide a GSE instead of .TS as input and at the output I would like to get the plain bb frames.
Something like:
`./leandvb ..... < capture.iq > out_bb_frame `

So basically bypass all the MPEG related processing.

Thank you

pabr, 2019-03-16:
Add this line, then run leandvb ... 4> out_bb_frame

```
@@ -680,6 +680,7 @@ int run_dvbs2(config &cfg) {
 
   unsigned long BUF_FRAMES = cfg.buf_factor;
   pipebuf<bbframe> p_bbframes(run.sch, "BB frames", BUF_FRAMES);
+  file_writer<bbframe> r_bbout(run.sch, p_bbframes, 4);
```

andimik, 2019-03-16:
BTW, does MIS (Multi input stream) also work?

pabr, 2019-03-16:
Neither GSE nor MIS are supported by leandvb yet. I am not aware of any practical use case.
Should be easy to implement on the RX side, though. TX is trickier because of realtime constraints.

andimik, 2019-03-16:
The practical use case is reception of Norwegian DAB radio via satellite.

See https://github.com/piratfm/eti-tools/blob/master/README.md#satellite-dab-feeds

pabr, 2019-03-16:
Thanks for the pointer. Sounds like something I might try someday, unless this satellite has a really tight beam that covers only Norway. SR 4800 will need a fast desktop PC, though.

andimik, 2019-03-16:
No, I am living in Austria next to the Slovenian border and the beam works here with an 80 cm dish...

I tried some DVB hardware (from Manufacturer TBS) but they cannot handle GSE.

cgavrila1, 2019-03-22:
Hi again,

Thank you very much for your help.
As mentioned, I'm tryin to transmit GSE instead of TS. 
I've added the above mentioned line and I was able to get the bb_frames. However, I'm facing an issue: the output bb_frames seem to contain some extra bytes (8 extra bytes - 1st frame; 10 extra bytes others )

Here is a screenshot with hex output of the 1st bb_frame (left: actual; right: expected):
[image]

(extra 8 bytes)

And a screenshot for the 2nd frame:
[image]

(extra 10 bytes)

Do you have any idea what I have to adapt?
Also i've attached 2 files with the actual and expected bb_frames. Please ignore the fact that they have different sizes, capturing duration was different, but the data is the same.

[extra_bytes_bb_frame.zip](https://github.com/pabr/leansdr/files/2996604/extra_bytes_bb_frame.zip)

I know that this is much to ask but would it be possible for you to take a look?
Many thanks

andimik, 2019-03-25:
Would this help? Recorded with hackrf. u8, SR 8MSPS

[image]

https://we.tl/t-mxUCdQQGJq
(90 MB)

pabr, 2019-03-26:
@cgavrila1 Yes there is a header, see struct bbframe and struct s2_pls. Should be fixed-size though. Ignore it if you already know your MODCOD and frame size.

cgavrila1, 2019-03-26:
Ok, thank you

pabr, 2019-03-26:
@andimik I believe your IQ waveform is saturated. You would get better results with reduced gain. Anyway you can get a few error-free bbframes with:
```
leandvb -f 8e6 --sr 4800e3 --standard DVB-S2 --sampler rrc --rrc-rej 30 --ldpc-helper ldpc_tool --gui -v -d < /tmp/SDRSharp_20190325_193654Z_10719425500Hz_IQ.wav  4> /tmp/nrk.out
```
(See www.pabr.org/radio/leandvb about ldpc_tool.)

"Unsupported or corrupted MODCOD" is caused by filler frames, a feature I hadn't encountered yet. Will fix ASAP to avoid unnecessary unlocking.

andimik, 2019-03-26:
Thanks for the hint. Your syntax described in the website was not clear to me regarding `--ldpc-helper COMMAND`

pabr, 2019-03-26:
@andimik I have added enough GSE support to process your file. Try:
```
leandvb -f 8e6  --sr 4800e3  --sampler rrc  --rrc-rej 30  --standard DVB-S2  --ldpc-helper ldpc_tool  --gui -v -d -d --fd-gse 3 < /tmp/SDRSharp_20190325_193654Z_10719425500Hz_IQ.wav 3> /tmp/nrk.out
```
Despite the saturation I get 240 kB of data. I can't confirm whether it is valid DAB/GSE, but there is definitely some Norwegian text in there.

This interface is for experimentation only; there is no provision for selecting a specific ISI nor signalling discontinuities in the outpout stream.

pabr, 2019-03-26:
To clarify: leandvb (work branch) can now handle the DVB-S2 features needed for GSE and output raw streams, but at this point you still have to process the GSE header yourself.

pabr, 2022-11-28:
@Dorfmaan No, MIS is not supported. You will see "MIS" instead of "SIS" in the debug messages ("leandvb -d -d") but the output will be garbage unless the modulator was careful about packing and padding payload data in bbframes.
If you have a practical use case and IQ recording, feel free to share.

Dorfmaan, 2022-11-29:
@pabr Yes, I'm trying to extract valid BBframes from this file: input.wav
Record parameters --s16 -f 4e6  --sr 1500e3 8PSK 2/3

The command is:
`./leandvb --s16 -f 4e6  --sr 1500e3 --standard DVB-S2 --modcods 13 --sampler rrc  --rrc-rej 30 --ldpc-helper ldpc_tool --gui -v -d -d --fd-gse 3 < /tmp/input.wav 3> /tmp/output.bb`

At the end give me:   ` .BB frames:    0/   0      4 writable  ,      0 unread ( 0 )`

Did I miss something?

Link for file: [https://www.transferxl.com/download/04jdPW8h7pLgRk](url)

pabr, 2022-11-29:
--modcods specifies a bitmask, not a modcod number.  The following demodulates 4.7 MiB of BB data for me.  But currently there is no option to filter either of the two streams from your recording.  The output does not include the ISI numbers either.  
`leandvb  --gui  --s16  --standard DVB-S2  --sr 1500e3  -f 4e6  -v -d -d  --ldpc-helper ldpc_tool  --fastlock  --sampler rrc --modcods 8192  --framesizes 1  --fd-gse 3  < input.wav  3> /tmp/output.bb`

andimik, 2022-11-29:
@Dorfmaan 

Two MIS and SR 1500? Seems you want to extract German DAB+ from 23.5E?

If yes: There is no need to extract one of the MIS. You just need https://github.com/newspaperman/bbframe-tools to do the rest.

But currently I cannot use `leansdr` resp `ldpc_tool` on my new laptop and have to re-install everything.

So please use the hint from @pabr and use the `output.bb` as input file and try to extract the two Bundesmuxes.

Dorfmaan, 2022-11-29:
@pabr Very strange. When I run the command: `leandvb --gui --s16 --standard DVB-S2 --sr 1500e3 -f 4e6 -v -d -d --ldpc-helper ldpc_tool --fastlock --sampler rrc --modcods 8192 --framesizes 1 --fd-gse 3 < input.wav 3> /tmp/output.bb`
I get an error message: `BCHCORR = 0
_partial read(LDPC helper): Resource temporarily unavailable`

@andimik Exactly! But unfortunately I get an error message.

@pabr Can you share with us the `output.bb` file?

pabr, 2022-11-30:
@Dorfmaan Does it fail immediately, or after decoding some frames ? Can you post the full stderr log ?

@andimik Sorry, I won't redistribute content of unknown origin, but here are the ISIs and a few ASCII strings from the payload. If I understand correctly, bbfudpdecap from bbframe-tools would extract DAB from these raw bbframes ?
```
BBH: crc 2a/2a(OK) GC MIS(ISI=1) ACM ro=0.20 upl=0 dfl=42960 sync=00 syncd=0
BBH: crc ef/ef(OK) GC MIS(ISI=2) ACM ro=0.20 upl=0 dfl=42960 sync=00 syncd=0
BBH: crc 26/26(OK) GC MIS(ISI=1) ACM ro=0.20 upl=0 dfl=31184 sync=00 syncd=0
BBH: crc 24/24(OK) GC MIS(ISI=2) ACM ro=0.20 upl=0 dfl=42776 sync=00 syncd=0
```
```
crid://dradio.de/dlf/1141721
KLASSIK RADIO
ANTENNE BAYERN
Berichte, Meinungen, Rezensionen
```

andimik, 2022-11-30:
Sorry, I told you the wrong tool

`cat output.bb | bbfedi2eti -dst-ip 239.128.72.10 -dst-port 50010 2> /dev/null | dablin_gtk`

from https://github.com/piratfm/eti-tools should work

Dorfmaan, 2022-11-30:
@pabr OK, I see my error and correct it with the command: `echo 2073600 > /proc/sys/fs/pipe-max-size` 
And one final question, which is very important for me and I don't understand. How do you calculate the exact value of `--modcods` bitmask value?

pabr, 2022-11-30:
@Dorfmaan For modcod 13 the bitmask is 2 to the power 13 (2^13, 2**13, 1<<13 depending on language), hence 8192.

@andimik With a few patches I got bbfedi2eti to scan through GSE headers inside the bbframes but it never reaches HandleEDIPacket(). I'd need to learn more about the GSE and EDI and the fragmentation scheme to debug further.
```
[11/30/22 13:30:08] EDI: stats: packets lost: 0/0 errors:0/0
```

pabr, 2022-11-30:
@andimik I figured the problem was the IP addresses, it works now.
```
DR Deutschland   (0x10bc)
 0 : KLASSIK RADIO    (0xd75b) Pri subch= 6 start=192 CUs= 54 PL=eep-3a bitrate=72 DAB+
 1 : sunshine live    (0x15dc) Pri subch=21 start=330 CUs= 54 PL=eep-3a bitrate=72 DAB+
 2 : radio horeb      (0xd01c) Pri subch= 5 start=156 CUs= 36 PL=eep-3a bitrate=48 DAB+
 3 : ERF Plus         (0x1a64) Pri subch= 2 start=  0 CUs= 48 PL=eep-3a bitrate=64 DAB+
 4 : Dlf              (0xd210) Pri subch=10 start=480 CUs=104 PL=eep-2a bitrate=104 DAB+
 5 : Dlf Kultur       (0xd220) Pri subch=11 start=584 CUs=112 PL=eep-2a bitrate=112 DAB+
 6 : Dlf Nova         (0xd230) Pri subch=12 start=696 CUs=104 PL=eep-2a bitrate=104 DAB+
 7 : DRadio DokDeb    (0xd240) Pri subch=13 start=800 CUs= 36 PL=eep-3a bitrate=48 DAB+
 8 : Schwarzwaldradio (0x100d) Pri subch= 7 start=246 CUs= 42 PL=eep-2b bitrate=64 DAB+
 9 : ENERGY           (0x1a45) Pri subch= 4 start=102 CUs= 54 PL=eep-3a bitrate=72 DAB+
10 : SCHLAGERPARADIES (0x10c3) Pri subch=20 start=288 CUs= 42 PL=eep-2b bitrate=64 DAB+
11 : Absolut relax    (0x17fa) Pri subch= 3 start= 48 CUs= 54 PL=eep-3a bitrate=72 DAB+
12 : RADIO BOB!       (0x15dd) Pri subch=22 start=384 CUs= 54 PL=eep-3a bitrate=72 DAB+
```

andimik, 2022-11-30:
Oh, great, this is a big success! Thanks!

pabr, 2022-12-01:
I pushed the required changes. Note --fd-bbf and --bbf-l3sync below.
```
leandvb  --gui  --s16  --standard DVB-S2  --sr 1500e3  -f 4e6  -v -d -d  --ldpc-helper ldpc_tool  --fastlock  --sampler rrc  --roll-off 0.2  --modcods 8192  --framesizes 1  --fd-bbf 3  --bbf-l3sync  < input.wav  3> /tmp/output.bb
bbfedi2eti  -mis 1  < /tmp/output.bb  2> /dev/null  |  ni2out --list
```

Dorfmaan, 2022-12-01:
@pabr Amazing. Excellent work!  `--bbf-l3sync` was the missing function. Works perfectly!

Dorfmaan, 2022-12-01:
@pabr  Amazing. Excellent work! `--bbf-l3sync` - this was the missing function, till now. Works perfectly!

andimik, 2022-12-01:
[image]

Yes, this is excellent work.