pabr/leansdr #18: VLC not open when try  ./leandvb  --gui  |  vlc -    Help

khanfar, 2019-05-08:
please help . 
when i try to run command for live view :
rtl_sdr  -f 742450000  -s 2400000  -g 10  -  |  ./leandvb  --gui  |  vlc - 

rtl_sdr is run and lean is run ok BUT VLC NOT RUN ! please help

khanfar, 2019-05-09:
i try to run lean with ldpc helper by this command :

rtl_sdr  -f 742500000  -s 2400000  -g 10  -  |  ./leandvb  --gui  --sr 2000e3  --sampler rrc  --rrc-rej 30  --standard DVB-S2  --ldpc-helper ldpc_tool  -v  -d  |  vlc - 

but it start few seconds and exit ! i got this in terminal : 

mwk@mwk-desktop:~/leansdr/src/apps$ rtl_sdr  -f 742500000  -s 2400000  -g 10  -  |  ./leandvb  --gui  --sr 2000e3  --sampler rrc  --rrc-rej 30  --standard DVB-S2  --ldpc-helper ldpc_tool  -v  -d  |  vlc - 
VLC media player 3.0.4 Vetinari (revision 3.0.4-0-gf615db6332)
ANF is disabled (requires a clean signal).
Measuring SPD
Roll-off 0.35
Fs after resampling/decimation: 2400000.000000 Hz
RRC interpolator: 53 steps
RRC filter: 497 coeffs.
Found 1 device(s):
  0:  Realtek, RTL284UHIDIR, SN: Khanfar1/40

Using device 0: Generic RTL2832U OEM
Output:
  '_': S2 frame received without errors
  '.': error-corrected S2 frame
  '!': S2 frame with remaining errors
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 742500000 Hz.
Tuner gain set to 8.70 dB.
Reading samples in async mode...
COARSE(50): -0.133088 rad/symb (-42363 Hz at 2000000 baud)
Ignoring coarse det, using 0.000000
ACQ
LOCKED
ACQ
LOCKED
ACQ
LOCKED
Creating LUT for QPSK ratecode 1
Spawning LDPC helper: modcod=6 sf=0
____Start TS at 105
Signal caught, exiting!
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
mwk@mwk-desktop:~/leansdr/src/apps$

khanfar, 2019-05-09:
when i try to run command for live view :
rtl_sdr -f 742450000 -s 2400000 -g 10 - | ./leandvb --gui | vlc -

rtl_sdr is run and lean is run ok BUT VLC NOT RUN

khanfar, 2019-05-09:
this command i use for decoding .raw files : 

mwk@mwk-desktop:~/leansdr/src/apps$ ./leandvb  --f32  -f 2400e3  --sr 2000e3  --sampler rrc  --rrc-rej 30  --standard DVB-S2  --ldpc-helper ldpc_tool  -v  -d  --gui < /tmp/gqrx_20190503_212107_742493900_2400000_fc.raw  > /tmp/video.ts

to open video : mplayer /tmp/video.ts

khanfar, 2019-05-09:
the --sr 2000e3 should change depend of symbol rate you captured and you want to decoding 
like --sr 500 or --sr333 .....

pabr, 2019-05-09:
Try offline first: Record to a file with rtl_sdr, then demodulate with leandvb, then decode with vlc. Same as with GQRX recordings, except without the "--f32" option. If you don't want to enable AGC on the rtl_sdr, use "leandvb --gui" to check that the signal is neither too weak nor saturated. Tweak the gain and frequency offset until it works.

For realtime mode you should add "--inpipe 32000000 --nhelpers 6" to the leandvb command line. Watch for warnings about configuring /proc/sys/fs/pipe-max-size. This should prevent "Short write, samples lost, exiting!".

If this still doesn't work then maybe your computer is simply too slow. Check whether any process is keeping a whole CPU core busy.

khanfar, 2019-05-09:
i try this like what you say : 
rtl_sdr  -f 742500000  -s 2400000  -g 10  -  |  ./leandvb  --gui  --inpipe 32000000  --nhelpers 6  --sr 2000e3  --sampler rrc  --rrc-rej 30  --standard DVB-S2  --ldpc-helper ldpc_tool  -v  -d  |  vlc -

but got this : 

VLC media player 3.0.4 Vetinari (revision 3.0.4-0-gf615db6332)
--inpipe: Operation not permitted
Try 'echo 32000000 > /proc/sys/fs/pipe-max-size'
Found 1 device(s):
  0:  Realtek, RTL284UHIDIR, SN: Khanfar1/40

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 742500000 Hz.
Tuner gain set to 8.70 dB.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
mwk@mwk-desktop:~/leansdr/src/apps$

khanfar, 2019-05-09:
then i try this : 
sudo echo 32000000 > /proc/sys/fs/pipe-max-size
but got this : 
bash: /proc/sys/fs/pipe-max-size: Permission denied

khanfar, 2019-05-09:
then i try this : 
sudo sysctl fs.pipe-max-size=4194304
and i got this : 
fs.pipe-max-size = 4194304

but still the same error like up

my pc ic i3 third generation 8gb of ram , i sucsess decoding from recording .raw file from gqrx then decoding it and use mplayer to view , but im looking now for LIVE decoding , any wrong in my setup ?

pabr, 2019-05-09:
"sudo echo 32000000 > /proc/sys/fs/pipe-max-size" will not work because ">" is evaluated before sudo.

Try "sudo  /bin/bash" then "echo 32000000 > /proc/sys/fs/pipe-max-size".
Or "sudo sysctl fs.pipe-max-size=32000000".
Or "leandvb --inpipe 4194304" if that's really the highest value your OS will accept for pipe-max-size.

To determine whether your system is fast enough, simply measure how long leandvb takes to process a gqrx file (with "--nhelpers 6"), and check whether that's faster than the duration of the recording.
Even with a fast PC you will need to get --inpipe working for realtime LDPC decoding.

andimik, 2019-05-09:
You have to be root for that. `sudo` is not sufficient here.

khanfar, 2019-05-09:
i follow this : 
open terminal : 
sudo /bin/bash
echo 32000000 > /proc/sys/fs/pipe-max-size
it seems ok but : 

then try this : open terminal in this location : ~/leansdr/src/apps/ and try this : 
 
rtl_sdr -f 742500000 -s 2400000 -g 10 - | ./leandvb --gui --inpipe 32000000 --nhelpers 6 --sr 2000e3 --sampler rrc --rrc-rej 30 --standard DVB-S2 --ldpc-helper ldpc_tool -v -d | vlc -

i got this :  

VLC media player 3.0.4 Vetinari (revision 3.0.4-0-gf615db6332)
ANF is disabled (requires a clean signal).
Measuring SPD
Roll-off 0.35
Fs after resampling/decimation: 2400000.000000 Hz
RRC interpolator: 53 steps
RRC filter: 497 coeffs.
Found 1 device(s):
  0:  Realtek, RTL284UHIDIR, SN: Khanfar1/40

Using device 0: Generic RTL2832U OEM
Output:
  '_': S2 frame received without errors
  '.': error-corrected S2 frame
  '!': S2 frame with remaining errors
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 742500000 Hz.
Tuner gain set to 8.70 dB.
Reading samples in async mode...
COARSE(50): -0.150370 rad/symb (-47864 Hz at 2000000 baud)
Ignoring coarse det, using 0.000000
ACQ
LOCKED
ACQ
LOCKED
Creating LUT for QPSK ratecode 1
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
____Start TS at 41
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
mwk@mwk-desktop:~/leansdr/src/apps$

khanfar, 2019-05-09:
the lean waterfall start 5 seconds and i see 4 dots then exit !

khanfar, 2019-05-09:
look at this youtube video , its works good when i record .raw file and decoding it (not live decoding ) but the problem in live decoding , 

https://www.youtube.com/watch?v=aGgaCsdAP8U

https://www.rtl-sdr.com/receiving-eshail-2-dvb-s2-on-ubuntu-with-leandvb/

i use this command to decoding offline .RAW file from gqrx , and it works very well like youtube video i post . i move the .raw file to tmp folder then decode it by this command  : 

mwk@mwk-desktop:~/leansdr/src/apps$ ./leandvb  --f32  -f 2400e3  --sr 2000e3  --sampler rrc  --rrc-rej 30  --standard DVB-S2  --ldpc-helper ldpc_tool  -v  -d  --gui < /tmp/gqrx_20190503_212107_742493900_2400000_fc.raw  > /tmp/video.ts

khanfar, 2019-05-09:
but the problem in live decoding , what im missing ?

khanfar, 2019-05-09:
i try to change the symbol rate --sr 333e3 

rtl_sdr -f 747120000 -s 2400000 -g 20 - | ./leandvb --gui --inpipe 32000000 --nhelpers 6 --sr 333e3 --sampler rrc --rrc-rej 30 --standard DVB-S2 --ldpc-helper ldpc_tool -v -d | vlc -

i got this :  

VLC media player 3.0.4 Vetinari (revision 3.0.4-0-gf615db6332)
ANF is disabled (requires a clean signal).
Measuring SPD
Roll-off 0.35
Fs after resampling/decimation: 2400000.000000 Hz
RRC interpolator: 8 steps
RRC filter: 451 coeffs.
Found 1 device(s):
  0:  Realtek, RTL284UHIDIR, SN: Khanfar1/40

Using device 0: Generic RTL2832U OEM
Output:
  '_': S2 frame received without errors
  '.': error-corrected S2 frame
  '!': S2 frame with remaining errors
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 747120000 Hz.
Tuner gain set to 19.70 dB.
Reading samples in async mode...
COARSE(50): 0.110325 rad/symb (5847 Hz at 333000 baud)
Ignoring coarse det, using 0.000000
ACQ
LOCKED
Creating LUT for QPSK ratecode 3
Spawning LDPC helper: modcod=7 sf=0
Spawning LDPC helper: modcod=7 sf=0
Spawning LDPC helper: modcod=7 sf=0
Spawning LDPC helper: modcod=7 sf=0
Spawning LDPC helper: modcod=7 sf=0
Spawning LDPC helper: modcod=7 sf=0
ACQ
LOCKED
ACQ
LOCKED
decoder failed at converging to a code word!
____Start TS at 46
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
mwk@mwk-desktop:~/leansdr/src/apps$

pabr, 2019-05-10:
Try rtl_sdr+leandvb live without vlc. If this works then maybe the problem is in vlc, thought that would be surprising.
> rtl_sdr -f 742500000 -s 2400000 -g 10 - | ./leandvb --gui --inpipe 32000000 --nhelpers 6 --sr 2000e3 --sampler rrc --rrc-rej 30 --standard DVB-S2 --ldpc-helper ldpc_tool -v -d > /tmp/stream.ts
> vlc /tmp/stream.ts

Also you can run "leandvb -d -d" for more debug messages.

khanfar, 2019-05-10:
i test this , its record the file in tmp folder see this : 

mwk@mwk-desktop:~/leansdr/src/apps$ rtl_sdr -f 742500000 -s 2400000 -g 10 - | ./leandvb --gui --inpipe 32000000 --nhelpers 6 --sr 2000e3 --sampler rrc --rrc-rej 30 --standard DVB-S2 --ldpc-helper ldpc_tool -v -d > /tmp/stream.ts
ANF is disabled (requires a clean signal).
Measuring SPD
Roll-off 0.35
Fs after resampling/decimation: 2400000.000000 Hz
RRC interpolator: 53 steps
RRC filter: 497 coeffs.
Found 1 device(s):
  0:  Realtek, RTL284UHIDIR, SN: Khanfar1/40

Using device 0: Generic RTL2832U OEM
Output:
  '_': S2 frame received without errors
  '.': error-corrected S2 frame
  '!': S2 frame with remaining errors
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 742500000 Hz.
Tuner gain set to 8.70 dB.
Reading samples in async mode...
COARSE(50): -0.155223 rad/symb (-49409 Hz at 2000000 baud)
Ignoring coarse det, using 0.000000
ACQ
LOCKED
Creating LUT for QPSK ratecode 1
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
Spawning LDPC helper: modcod=6 sf=0
____Start TS at 143
________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________ACQ
LOCKED
ACQ
LOCKED
____________________________________ACQ
LOCKED
ACQ
LOCKED
ACQ
LOCKED
________ACQ
LOCKED
ACQ
LOCKED
____________________ACQ
LOCKED
ACQ
LOCKED
________________________________________________________________________________________________________________________________ACQ
LOCKED
ACQ
LOCKED
ACQ
LOCKED
____________________________________________________________________ACQ
LOCKED
____________________________________________________________________________________________________________________________________Lost a bbframe ?
Start TS at 97
________________________________ACQ
LOCKED
ACQ
LOCKED
________________________________Lost a bbframe ?
____Start TS at 67
Lost a bbframe ?
[... 379 lines cut ...]
________________________________________________________________Lost a bbframe ?
Start TS at 79
________________________________ACQ
LOCKED
ACQ
LOCKED
________________________________Lost a bbframe ?
Start TS at 167
____________________________Lost a bbframe ?
____Start TS at 87
________________________________________________________________________________________________ACQ
LOCKED
ACQ
LOCKED
________________________________________________________Lost a bbframe ?
____Start TS at 81
____________________________Lost a bbframe ?
____Start TS at 161
____________________________________________________________________________________________________^CSignal caught, exiting!
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...

mwk@mwk-desktop:~/leansdr/src/apps$

khanfar, 2019-05-10:
to exit i type ctrl+c , but when i try : 
vlc /tmp/stream.ts

it give me this : 
VLC media player 3.0.4 Vetinari (revision 3.0.4-0-gf615db6332)

the vlc not start , so the problem in vlc itself ? 
i try this : 
mplayer /tmp/stream.ts

its ok , it run in mplayer , but the problem i cant let the stream run for like 1/2 hours ! it give me big file save to tmp folder ! how to fix now vlc and not recording .ts to tmp ? , the issue i cant record big file , i need nethod for live , how to fix or reinstall vlc recommendation steps ? or use different player ?

pabr, 2019-05-10:
Yes, it looks like VLC does not like the TS.

I have successfully used mplayer instead for live decoding, e.g.:
> rtl_sdr ... | leandvb ... | mplayer -
or
> rtl_sdr ... | leandvb ... | mplayer -demuxer mpegts -tsprobe 1 -vid 256 -aid 257 -

khanfar, 2019-05-10:
Pabr ... its work now ,,, wow , thanks .... i reinstall vlc and type this : 

rtl_sdr -f 742500000 -s 2400000 -g 10 - | ./leandvb --gui --inpipe 32000000 --nhelpers 6 --sr 2000e3 --sampler rrc --rrc-rej 30 --standard DVB-S2 --ldpc-helper ldpc_tool -v -d | vlc -
its work now live decoding , but i got little lag , i check my cpu its 100% load , any thing to do to reduce the lag or make specific time like 10 seconds of buffering to catcg good live view ????

khanfar, 2019-05-10:
my sdr need little ppm correction from real frequency , how to add ppm correction value ?? and how to make the symbole rate in lean auto detected so i put now --sr 2000e3 for 2mS/s , can i make the lean auto detect the Symbol rate  ???

khanfar, 2019-05-10:
were to add these commands ?

I have successfully used mplayer instead for live decoding, e.g.:

    rtl_sdr ... | leandvb ... | mplayer -
    or
    rtl_sdr ... | leandvb ... | mplayer -demuxer mpegts -tsprobe 1 -vid 256 -aid 257 -

khanfar, 2019-05-11:
i restart the pc and run the command again : 

rtl_sdr -f 742500000 -s 2400000 -g 10 - | ./leandvb --gui --inpipe 32000000 --nhelpers 6 --sr 2000e3 --sampler rrc --rrc-rej 30 --standard DVB-S2 --ldpc-helper ldpc_tool -v -d | vlc -

then i got this error : 

Your input can't be opened:
VLC is unable to open the MRL 'fd://0'. Check the log for details.

khanfar, 2019-05-11:
--inpipe: Operation not permitted
Try 'echo 32000000 > /proc/sys/fs/pipe-max-size'
VLC media player 3.0.4 Vetinari (revision 3.0.4-0-gf615db6332)
Found 1 device(s):
  0:  Realtek, RTL284UHIDIR, SN: Khanfar1/40

Using device 0: Generic RTL2832U OEM
[00005617d695a570] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
Found Rafael Micro R820T tuner
XmbTextListToTextProperty result code -2
[00007fe6ac017c80] mjpeg demux error: cannot peek
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 742500000 Hz.
Tuner gain set to 8.70 dB.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
QObject::~QObject: Timers cannot be stopped from another thread
mwk@mwk-desktop:~/leansdr/src/apps$

pabr, 2019-05-11:
On lag: There is no easy way to reduce latency. It is caused by large DVB-S2 frames, buffering for SIMD processing, and LDPC decoding jitter at low SNR. Try other values for "--nhelpers", for example "--nhelpers 3" if you CPU has 4 cores.

On ppm correction: You can try "rtl_sdr -p", or adjust the value directly in "rtl_sdr -f", or use "leandvb --derotate".
 
On symbol rate: leandvb does not support auto-detection. It ships with "leansdrscan" which is intended to cycle through several combinations of frequency and symbol rates, but I haven't tested it in a while.

On mplayer: You  can write "mplayer" instead of "vlc" in your commands. Sometimes it works better, sometimes not.

khanfar, 2019-05-11:
the main frequency i need to decoding exactly REAL  748.750 but when i put the command rtl_sdr -f 748.750 the waterfall in lean not center in REAL frequency so i put the frequency -f 748.610 to got it center to real frequency ,,, so the lean decoding the main real frequency in my case in 748.610 , .....
i try now to put rtl_sdr -p 200 , so now in waterfall its center in lean ,,,, BUT the lean not decoding !!!!
what should i do in this case ??? whats about leandvb --derotate , and how to use it ? should i need to put this ? and whats the value ?

pabr, 2019-05-12:
The frequency error is -140 kHz, which is 187 ppm of 748.750 MHz, so you can try the following alternatives:

> rtl_sdr -f 748.610e6 ...  |  leandvb ...

> rtl_sdr -f 748.750e6 ...  |  leandvb --derotate -140e3 ...

> rtl_sdr -p 187 -f 748.750e6 ...  |  leandvb ...        (or maybe "-p -187")

Note: "leandvb --gui" tells you the residual frequency offset (blue text in timeline window). If it's more than +-10 kHz you can use that value to adjust the frequency correction and get better demodulation.

khanfar, 2019-05-12:
how you calculate the  187 ppm value ?? the frequency error is -140Kz , but how to calculate the ppm value = 187 ppm ?

khanfar, 2019-05-12:
is there way to adjust tune when lean run in real time ? i mean not exit the software and edit the command value like to correct the ppm or frequency error

pabr, 2019-05-14:
140 kHz divided by 748.750 MHz is 0.000187. That's 187 parts per million (ppm).

Note: If your rtl_dr is a high-end model (with a TCXO) then the 140 kHz error must come from the inaccuracy of the LNB oscillator, not from the rtl_sdr itself. In that case you should not use "rtl_sdr -p", because this might also affect the sampling rate. Simply adjust the tuning frequency by 140 kHz.

There is no way to retune leandvb at runtime, but future version will tolerate larger frequency offsets.

khanfar, 2019-05-18:
thanks pabr fo your help ... we are looking for new version ,,, hope soon ,,,, and look for can adjust frequency realtime on terminal by +_ buttons from keyboard , or build a easy GUI to put frequency and gain and offset and symbol rate in GUI easy window , then click start to run the software ...... thanks