pabr/leansdr #20: leandvb closes unexpectedly after loosing sync

astrohead, 2019-07-25:
Hello,

I am trying to demodulate a DVBS .iq file (generated in GNUradio from a TS file source and recorded with a USRP). Once I launch leandvb, it exits before the file is finished and it is not able to correctly find the SYNC byte in the TS stream.

I attach the output in the terminal and the GUI.

[image]

`./leandvb --cr 1/2  --f32  -f 1e6  -d --sr 500e3  --gui < /home/myRecording.iq 
deconv[0]=0x00000000000003ba 7 taps / 10 bits
Output:
  '_': TS packet received without errors
  '.': error-corrected TS packet
  '!': TS packet with remaining errors
Locked
!!!!(ed)!!!!derandomizer: resynchronizing
!!!!(19)!!!!(8f)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(14)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(08)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!Unlocked
!!!!(72)!!!!!!!!!!!!!!!!Locked
!!!!!!!!!!!!!!!!!!!!!!!!(54)!!!!(82)(1f)!!!!derandomizer: resynchronizing
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(0f)(22)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(31)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(ec)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(0d)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(a0)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(9b)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(a2)!!!!!!!!!!!!!!!!!!!!!!!!(bd)!!!!!!!!!!!!!!!!!!!!!!!!(51)!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!(92)!!!!!!!!!!!!!!!!!!!!!!!!
.rawiq            :  85M/ 14M  10403 writable  ,   3037 unread ( 989 3037 989 989 3037 93 )
.stdin            :  14M/ 14M    675 writable  ,      0 unread ( 0 )
.cnr              :    0/   0      4 writable  ,      0 unread ( )
.spectrum         :    0/  14      2 writable  ,      0 unread ( )
.freq             :   71/  71      2 writable  ,      0 unread ( 0 )
.SS               :   71/  71      2 writable  ,      0 unread ( 0 )
.MER              :   71/  71      2 writable  ,      0 unread ( 0 )
.cstln            : 110k/111k  12489 writable  ,    823 unread ( 823 )
.PLS cstln        :    0/   0  16384 writable  ,      0 unread ( )
.frame lock       :    0/   0      4 writable  ,      0 unread ( )
.VBER             :    0/   0      4 writable  ,      0 unread ( )
.lock             :    0/   4      4 writable  ,      0 unread ( )
.locktime         :    0/4071    141 writable  ,      0 unread ( )
.TS packets       :    0/   0    156 writable  ,      0 unread ( 0 0 )
.Bits processed   : 1015/1015     77 writable  ,      0 unread ( 0 )
.Bits corrected   : 1015/1015     77 writable  ,      0 unread ( 0 )
.VBER             :    0/ 126      2 writable  ,      0 unread ( )
.packet counter   :    0/   0    156 writable  ,      0 unread ( 0 )
.PSK soft-symbols :   7M/  7M   3674 writable  ,     70 unread ( 70 )
.bytes            : 891k/891k    868 writable  ,    184 unread ( 184 )
.mpegbytes        : 828k/830k    204 writable !,   2244 unread ( 2244 )
.RS-enc packets   : 4060/4060      0 writable !,      0 unread ( 0 )
.rand TS packets  : 4060/4060      0 writable !,      0 unread ( 0 )
Total buffer memory: 770 KiB`

Thank you in advance!

pabr, 2019-07-25:
Try adding "--sampler rrc" and/or "--viterbi" to increase sensitivity.
Also, if possible, try recording with a higher sampling rate (e.g. 4 IQ samples per symbol instead of 2).

astrohead, 2019-07-26:
Thank you for your reply. 

After increasing the sampling rate in the recording by a factor x2, activating Viterbi and the RRC filtering, I get the following output:

[image]

`./leandvb  --f32  -f 2e6 --viterbi --sampler rrc -d --sr 500e3 --gui < /home/myRecording.iq
Output:
  '_': TS packet received without errors
  '.': error-corrected TS packet
  '!': TS packet with remaining errors
{0->1}{1->3}{3->2}{2->0}{0->3}{3->2}{2->0}{0->3}{3->2}{2->0}{0->2}{2->3}{3->0}{0->2}{2->3}{3->1}{1->0}{0->1}{1->0}{0->1}{1->2}{2->1}{1->3}{3->0}{0->3}{3->0}{0->1}{1->3}{3->2}{2->3}{3->2}{2->0}{0->3}{3->2}{2->1}{1->0}{0->2}{2->3}{3->0}{0->3}{3->1}{1->2}{2->1}{1->3}{3->0}{0->1}{1->2}{2->0}{0->1}{1->2}{2->3}{3->1}{1->3}{3->2}{2->0}{0->1}{1->0}{0->2}{2->1}{1->2}{2->0}{0->1}{1->2}{2->1}{1->2}{2->0}{0->1}{1->3}{3->2}{2->3}{3->2}{2->0}{0->3}{3->2}{2->0}{0->2}{2->1}{1->2}{2->0}{0->1}{1->0}{0->3}{3->0}{0->2}{2->3}{3->0}{0->2}{2->1}{1->0}{0->2}{2->1}{1->3}{3->1}{1->3}{3->1}{1->3}{3->0}{0->1}{1->3}{3->2}{2->0}{0->1}{1->2}{2->0}{0->2}{2->3}{3->1}{1->3}{3->1}{1->0}{0->3}{3->0}{0->1}{1->3}{3->1}{1->0}{0->1}{1->3}{3->1}{1->0}{0->2}{2->3}{3->2}{2->1}{1->3}{3->1}{1->2}{2->0}{0->1}{1->0}{0->2}{2->3}{3->0}{0->2}{2->0}{0->2}{2->0}{0->2}{2->1}{1->2}{2->0}{0->3}{3->1}{1->3}{3->0}{0->2}{2->0}{0->1}{1->3}{3->0}{0->3}{3->1}{1->0}{0->2}{2->3}{3->0}{0->1}{1->0}{0->1}{1->0}{0->3}{3->0}{0->1}{1->0}{0->3}{3->0}{0->1}{1->0}{0->1}{1->2}{2->1}{1->3}{3->2}{2->3}{3->1}{1->3}{3->2}{2->1}{1->2}{2->0}{0->1}{1->2}{2->0}{0->1}{1->0}{0->1}{1->0}{0->2}{2->3}{3->0}{0->2}{2->1}{1->2}{2->1}{1->0}{0->2}{2->0}{0->1}{1->3}{3->0}{0->2}{2->1}{1->2}{2->0}{0->2}{2->3}{3->2}{2->0}{0->1}{1->2}{2->1}{1->0}{0->1}{1->0}{0->3}{3->2}{2->0}{0->3}{3->1}{1->0}{0->3}{3->0}{0->1}{1->0}{0->1}{1->3}{3->0}{0->1}{1->2}{2->1}{1->2}{2->0}{0->1}{1->3}{3->1}{1->3}{3->2}{2->0}{0->2}{2->3}{3->0}{0->3}{3->2}{2->3}{3->1}{1->0}{0->2}{2->1}{1->2}{2->0}{0->1}{1->0}{0->1}{1->0}{0->2}{2->0}{0->3}{3->2}{2->1}{1->0}{0->3}{3->0}{0->2}{2->1}{1->3}{3->1}{1->3}{3->0}{0->2}{2->1}{1->0}{0->2}{2->1}{1->3}{3->1}{1->0}{0->3}{3->0}{0->3}{3->2}{2->0}{0->1}{1->2}{2->0}{0->3}{3->2}{2->1}{1->3}{3->2}{2->0}{0->3}{3->0}{0->1}{1->0}{0->3}{3->1}{1->3}{3->1}{1->0}{0->2}{2->1}{1->0}{0->1}{1->2}{2->1}{1->3}{3->2}{2->0}{0->1}{1->2}{2->1}{1->0}{0->1}{1->2}{2->1}{1->0}{0->3}{3->2}{2->1}{1->3}{3->1}{1->3}{3->1}{1->2}{2->3}{3->0}{0->1}{1->0}{0->1}{1->0}{0->1}{1->2}{2->0}{0->3}{3->0}{0->1}{1->2}{2->3}{3->1}{1->0}{0->1}{1->0}{0->1}{1->3}{3->1}{1->0}{0->2}{2->1}{1->0}{0->3}{3->1}{1->3}{3->1}{1->0}{0->1}{1->2}{2->3}{3->1}{1->2}{2->0}{0->1}{1->3}{3->2}{2->1}{1->3}{3->1}{1->2}{2->0}{0->2}{2->1}{1->3}{3->0}{0->2}{2->0}{0->1}{1->0}{0->1}{1->2}{2->0}{0->3}{3->2}{2->0}{0->2}{2->1}{1->3}{3->0}{0->1}{1->0}{0->1}{1->2}{2->0}{0->1}{1->3}{3->0}{0->3}{3->1}{1->3}{3->1}{1->0}{0->2}{2->1}{1->0}{0->2}{2->3}{3->2}{2->0}{0->1}{1->3}{3->0}{0->1}{1->3}{3->1}{1->3}{3->0}{0->2}{2->0}{0->2}{2->0}{0->2}{2->3}{3->1}{1->2}{2->0}{0->1}{1->3}{3->1}{1->0}{0->2}{2->3}{3->2}{2->0}{0->1}{1->0}{0->2}{2->0}{0->2}{2->1}{1->3}{3->1}{1->0}{0->3}{3->2}{2->3}{3->1}{1->0}{0->2}{2->0}{0->3}{3->0}{0->1}{1->0}{0->1}{1->0}{0->1}{1->2}{2->1}{1->3}{3->0}{0->3}{3->1}{1->0}{0->3}{3->1}{1->3}{3->2}{2->1}{1->0}{0->2}{2->3}{3->2}{2->0}{0->2}{2->3}{3->0}{0->2}{2->3}{3->1}{1->0}{0->3}{3->0}{0->1}{1->0}{0->1}{1->0}{0->2}{2->1}{1->2}{2->1}{1->3}{3->0}{0->3}{3->2}{2->1}{1->0}{0->1}{1->2}{2->1}{1->0}{0->2}{2->0}{0->3}{3->0}{0->3}{3->1}{1->2}{2->3}{3->2}{2->1}{1->3}{3->1}{1->0}{0->3}{3->0}{0->2}{2->3}{3->0}{0->2}{2->0}{0->2}{2->0}{0->3}{3->2}{2->3}{3->0}{0->1}{1->0}{0->2}{2->1}{1->3}{3->1}{1->0}{0->2}{2->0}{0->2}{2->1}{1->3}{3->0}{0->3}{3->1}{1->0}{0->2}{2->3}{3->0}{0->2}{2->0}{0->1}{1->0}{0->1}{1->2}{2->3}{3->0}{0->2}{2->1}{1->0}{0->1}{1->3}{3->2}{2->1}{1->0}{0->1}{1->0}{0->1}{1->0}{0->3}{3->0}{0->3}{3->0}{0->2}{2->1}{1->2}{2->1}{1->2}{2->1}{1->0}{0->1}{1->2}{2->3}{3->0}{0->3}{3->1}{1->3}{3->0}{0->2}{2->0}{0->3}{3->0}{0->1}{1->0}{0->2}{2->0}{0->2}{2->0}{0->1}{1->3}{3->0}{0->3}{3->1}{1->2}{2->3}{3->0}{0->3}{3->1}{1->2}{2->1}{1->0}{0->3}{3->1}{1->0}{0->2}{2->0}{0->1}{1->2}{2->3}{3->1}{1->2}{2->1}{1->3}{3->0}{0->3}{3->0}{0->1}{1->2}{2->1}{1->3}{3->2}{2->3}{3->1}{1->2}{2->3}{3->1}{1->3}{3->2}{2->0}{0->3}{3->1}{1->2}{2->1}{1->3}{3->0}{0->1}{1->0}{0->3}{3->1}{1->2}{2->0}{0->1}{1->0}{0->3}{3->0}{0->1}{1->2}{2->3}{3->2}{2->1}{1->3}{3->1}{1->0}{0->1}{1->3}{3->1}{1->3}{3->1}{1->2}{2->3}{3->0}{0->2}{2->1}{1->0}{0->2}{2->3}{3->1}{1->0}{0->2}{2->0}{0->2}{2->3}{3->1}{1->2}{2->1}{1->2}{2->0}{0->3}{3->2}{2->1}{1->0}{0->2}{2->3}{3->1}{1->0}{0->3}{3->2}{2->1}{1->0}{0->3}{3->2}{2->0}{0->1}{1->0}{0->1}{1->3}{3->1}{1->3}{3->1}{1->2}{2->1}{1->3}{3->2}{2->1}{1->2}{2->3}{3->1}{1->0}{0->2}{2->3}{3->1}{1->0}{0->3}{3->0}{0->2}{2->1}{1->0}{0->1}{1->3}{3->0}{0->1}{1->2}{2->0}{0->1}{1->2}{2->0}{0->2}{2->3}{3->2}{2->3}{3->2}{2->0}{0->2}{2->1}{1->3}{3->1}{1->0}{0->3}{3->1}{1->2}{2->0}{0->2}{2->0}{0->1}{1->0}{0->3}{3->2}{2->1}{1->2}{2->0}{0->2}{2->1}{1->3}{3->0}{0->3}{3->2}{2->0}{0->3}{3->0}{0->1}{1->0}{0->1}{1->2}{2->0}{0->3}{3->2}{2->1}{1->2}{2->1}{1->2}{2->0}{0->2}{2->1}{1->3}{3->0}{0->1}{1->0}{0->3}{3->0}{0->1}{1->0}{0->2}{2->1}{1->3}{3->0}{0->3}{3->0}{0->1}{1->0}{0->1}{1->3}{3->1}{1->3}{3->0}{0->1}{1->0}{0->2}{2->3}{3->0}{0->3}
.rawiq            : 108M/ 18M  14591 writable  ,   1537 unread ( 513 1537 513 513 1537 257 )
.stdin            :  18M/ 18M   2303 writable  ,      0 unread ( 0 )
.cnr              :    0/   0      4 writable  ,      0 unread ( )
[... cut ...]
.packet counter   :    0/   0    156 writable  ,      0 unread ( 0 )
.PSK soft-symbols :   4M/  4M   5693 writable  ,     67 unread ( 67 )
.bytes            : 566k/567k   1888 writable  ,   1408 unread ( 1408 )
.mpegbytes        :    0/   0   9792 writable  ,      0 unread ( 0 )
.RS-enc packets   :    0/   0      4 writable  ,      0 unread ( 0 )
.rand TS packets  :    0/   0      4 writable  ,      0 unread ( 0 )
Total buffer memory: 770 KiB`

How can I invert the DVB-S spectrum? I think this could eventually help me.

Thank you!

astrohead, 2019-07-26:
I have uploaded [here ](https://www.swisstransfer.com/d/f4efe0e4-34ea-4b2f-9852-28f8614c69f3) the DVBS .iq file in case someone wants to give it a try and demodulate it with leansdr.

Sampling rate: 1M
Symbol rate: 500K
Data format: Complex32
FEC: 1/2

pabr, 2019-07-26:
I can't demodulate it either, and it's hard to tell what is wrong. SNR is good but the constellation is skewed (diamond shape instead of square).

If I understand correctly this signal was generated by gnuradio and sent over the air to a USRP. Maybe you could set up gnuradio to output IQ data to a file and demodulate that. This may reveal whether the problem is a modulation compatibility issue or a RF issue.

You can invert the spectrum by negating the Q samples, but IIRC leandvb handles that automatically.

astrohead, 2019-07-27:
Hello,

Thank you for your answer. 

I have tried to emulate a DVB-S transmitter in GNURadio, record the IQ data and demodulate it with leandvb. This is working perfectly well!

However, the recording that I uploaded here comes from the output of a proprietary DVB-S modulator. I still cannot demodulate it with leandvb. What is strange, is that using an off-the-shelf DVB-S tuner ([TBS5520SE](https://www.tbsdtv.com/products/tbs5520se_multi-standard_tv_tuner_usb_box.html)) I can demodulate it without problems.

Is there a parameter that I might not be configuring correctly in leandvb? Or maybe leandvb is a less sensitive demodulator than the TBS5520SE?

Any suggestions or ideas are welcome.

Thank you in advance.