pabr/leansdr #21: Compile: libiio and X11 not found

OZ1SEJ, 2019-08-21:
When I compile leandvb, I get these notifications:

    libiio not found. Will not build leaniio{rx,tx}.
    X11 not found. Will not compile GUI.

How do I resolve these? (Ubuntu 19.04)

pabr, 2019-08-21:
You should now have a working "leandvb" binary anyway, except that it won't support the "--gui" command-line flag. For this you probably need to install a Ubuntu package called "libx11-dev" or something similar.

leaniiorx and leaniiotx are distinc applications, not required for leandvb.

pabr, 2019-08-21:
You should now have a working "leandvb" binary anyway, except that it won't support the "--gui" command-line flag. For this you probably need to install a Ubuntu package called "libx11-dev" or something similar.

leaniiorx and leaniiotx are distinct applications, not required for leandvb.

OZ1SEJ, 2019-08-21:
Ah, libx11-dev worked - thanks!

closed 2019-08-21 by OZ1SEJ