pabr/leansdr #22: Inefficiencies with variable MODCOD (VCM)

igorauad, 2019-11-19:
Hello all,

Today I've tried to take my first steps into leansdr. I was able to run it to decode a DVB-S2 signal that uses VCM, specifically with two multiplexed MODCOD streams that are associated to two distinct MPEG TS PIDs. Despited success on reception, I noticed there is a substantial jump in CPU usage when the two MODCODs are active. If, in contrast, I disable one MODCOD and leave only one active the CPU usage drops quite substantially.

I haven't had a chance to look at the implementation yet, but I suppose this is related to frame locking and particularly the different lengths of the PLFRAMEs of each MODCOD. 

So, before delving into this, I wanted to ask:
1) What is the cause of the problem? Is it really due to frame locking? 
2) Are there any potential solutions?

I'm interested in contributing with a solution.

Many thanks

pabr, 2019-11-19:
leandvb pre-computes data about constellations to speed-up demodulation, but currently it stores data for one constellation only. So when you feed it a VCM signal it recomputes this data everytime the constellation changes, i.e. possibly every frame, hence the dramatic CPU load.

There is no technical difficulty in fixing this; I just need a convincing use case where users really need to demodulate VCM streams in software. AFAIK VCM is associated with sophisticated applications (high bandwidth multiplexes and ACM on point-to-point links) where hardware demodulators would be preferred.

The relevant code is in s2_frame_receiver::run_frame_locked() where it prints "Warning: Variable MODCOD is inefficient".

Regarding the other question, filtering by MODCOD is not implemented. Since DVB-S2 does not tag logical streams at the PLFRAME level, this would not be a reliable way to filter a multiplex. But as a quick hack it could reduce CPU load by skipping LDPC+BCH error-correction on unwanted frames.

igorauad, 2019-11-20:
Thanks for the prompt response, @pabr .

Indeed it seems to be an easy fix. I've updated the pointer `cstln` to the soft-decoding look-up table. Now, instead of a pointer, I've made it an array of pointers with fixed space for 32 pointers, one for each possible MODCOD. So, when receiving a frame, I check if a look-up table already exists for that MODCOD and then I use it. By doing so, the LUT does not need to be recomputed every time.

Do you see any significant drawback on this other than using more memory?

I'm also storing the most recently received MODCOD as a member of the `s2_frame_receiver`, so that function `update_agc` can adjust the gain based on that. However, since the AGC adjusts gain before detecting the MODCOD of the incoming frame, my understanding is that this would often adjust based on the wrong MODCOD when MODCODs are multiplexed. Any suggestions?

pabr, 2019-11-20:
Might waste a lot of memory in the future with ACM on DVB-S2X (which has many more MODCODs), but that's what I would do for now.

The AGC issue arises only if VCM and --strongpls are used simultaneously. This is unlikely to happen in practice, so I wouldn't worry too much about that.