pabr/leansdr #23: libiio -- not found openSSE -- it is installed

KC7NOA, 2019-12-13:
probably not your concern

kc7noa@install:~/Downloads/leansdr/src/apps> make
libiio not found. Will not build leaniio{rx,tx}.
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS  leandvbtx.cc   -o leandvbtx
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS -DGUI leandvb.cc  -lX11 -o leandvb
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS  leansdrscan.cc   -o leansdrscan
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS  leansdrserv.cc   -o leansdrserv
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS  leantsgen.cc   -o leantsgen
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS  leansdrcat.cc   -o leansdrcat
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS  leanchansim.cc   -o leanchansim
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-87-g94e30b5\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable -DLEANSDR_EXTENSIONS  leanmlmrx.cc  -lfftw3f -lpthread -o leanmlmrx
kc7noa@install:~/Downloads/leansdr/src/apps> sudo make install
[sudo] password for root: 
libiio not found. Will not build leaniio{rx,tx}.
make: *** No rule to make target 'install'.  Stop.
kc7noa@install:~/Downloads/leansdr/src/apps>

pabr, 2019-12-13:
Simply copy the binaries you are interested in where you need them. There are no dynamic libraries, configuration files, man pages or other things to install.

I don't understand how this relates to openSSE.

KC7NOA, 2019-12-13:
I simply have openSUSE as my OS,

Im not sure where libiio is installed ... or if it is a binary when installed ...

I did find and installed libiio, python's libiio and libiio-devil with zypper se libiio

I hope to be using LimeSDR-USB and SdrAngel to capture DVB ... it will not be I/Q i think ...

pabr, 2019-12-13:
libiio is only required for PlutoSDR-specific tools. You don't need them if you intend to record files with SdrAngel and process them offline with leandvb.

KC7NOA, 2019-12-13:
I'd like to decode live stream .... if possible

Sent from my Verizon, Samsung Galaxy smartphone