pabr/leansdr #24: leansdr make embedded issue for plutosdr

khan2018, 2020-02-20:
Hi
I was installing leansdr and I included the binaries for libiio and the make step went through.

But when I do make embedded I get the following error.

$ make embedded
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-27-ge6416aa\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable  -Ilibiio leaniiorx.cc  -lfftw3f -lpthread -Llibiio -L. -liio -lpthread -lxml2 -lz -llzma -static -o leaniiorx.x86_64
/usr/bin/ld: cannot find -liio
collect2: error: ld returned 1 exit status
Makefile:83: recipe for target 'leaniiorx.x86_64' failed
make: *** [leaniiorx.x86_64] Error 1

Output of 
$ ld -liio --verbose

GNU ld (GNU Binutils for Ubuntu) 2.26.1
  Supported emulations:
   elf_x86_64
   elf32_x86_64
   elf_i386
   elf_iamcu
   i386linux
   elf_l1om
   elf_k1om
   i386pep
   i386pe
using internal linker script:
==================================================
/* Script for -z combreloc: combine and sort reloc sections */
/* Copyright (C) 2014-2015 Free Software Foundation, Inc.
   Copying and distribution of this script, with or without modification,
   are permitted in any medium without royalty provided the copyright
   notice and this notice are preserved.  */
OUTPUT_FORMAT("elf64-x86-64", "elf64-x86-64",
	      "elf64-x86-64")
OUTPUT_ARCH(i386:x86-64)
ENTRY(_start)
SEARCH_DIR("=/usr/local/lib/x86_64-linux-gnu"); SEARCH_DIR("=/lib/x86_64-linux-gnu"); SEARCH_DIR("=/usr/lib/x86_64-linux-gnu"); SEARCH_DIR("=/usr/local/lib64"); SEARCH_DIR("=/lib64"); SEARCH_DIR("=/usr/lib64"); SEARCH_DIR("=/usr/local/lib"); SEARCH_DIR("=/lib"); SEARCH_DIR("=/usr/lib"); SEARCH_DIR("=/usr/x86_64-linux-gnu/lib64"); SEARCH_DIR("=/usr/x86_64-linux-gnu/lib");
SECTIONS
{
  /* Read-only sections, merged into text segment: */
  PROVIDE (__executable_start = SEGMENT_START("text-segment", 0x400000)); . = SEGMENT_START("text-segment", 0x400000) + SIZEOF_HEADERS;
  .interp         : { *(.interp) }
  .note.gnu.build-id : { *(.note.gnu.build-id) }
  .hash           : { *(.hash) }
  .gnu.hash       : { *(.gnu.hash) }
  .dynsym         : { *(.dynsym) }
  .dynstr         : { *(.dynstr) }
  .gnu.version    : { *(.gnu.version) }
  .gnu.version_d  : { *(.gnu.version_d) }
  .gnu.version_r  : { *(.gnu.version_r) }
  .rela.dyn       :
    {
      *(.rela.init)
      *(.rela.text .rela.text.* .rela.gnu.linkonce.t.*)
      *(.rela.fini)
      *(.rela.rodata .rela.rodata.* .rela.gnu.linkonce.r.*)
      *(.rela.data .rela.data.* .rela.gnu.linkonce.d.*)
      *(.rela.tdata .rela.tdata.* .rela.gnu.linkonce.td.*)
      *(.rela.tbss .rela.tbss.* .rela.gnu.linkonce.tb.*)
      *(.rela.ctors)
      *(.rela.dtors)
      *(.rela.got)
      *(.rela.bss .rela.bss.* .rela.gnu.linkonce.b.*)
      *(.rela.ldata .rela.ldata.* .rela.gnu.linkonce.l.*)
      *(.rela.lbss .rela.lbss.* .rela.gnu.linkonce.lb.*)
      *(.rela.lrodata .rela.lrodata.* .rela.gnu.linkonce.lr.*)
      *(.rela.ifunc)
    }
  .rela.plt       :
[... 202 lines cut ...]
libpthread.so.0 needed by //usr/lib/x86_64-linux-gnu/libiio.so
found libpthread.so.0 at //lib/x86_64-linux-gnu/libpthread.so.0
libc.so.6 needed by //usr/lib/x86_64-linux-gnu/libiio.so
found libc.so.6 at //lib/x86_64-linux-gnu/libc.so.6
libudev.so.1 needed by //lib/x86_64-linux-gnu/libusb-1.0.so.0
found libudev.so.1 at //lib/x86_64-linux-gnu/libudev.so.1
libdl.so.2 needed by //usr/lib/x86_64-linux-gnu/libxml2.so.2
found libdl.so.2 at //lib/x86_64-linux-gnu/libdl.so.2
libicuuc.so.55 needed by //usr/lib/x86_64-linux-gnu/libxml2.so.2
found libicuuc.so.55 at //usr/lib/x86_64-linux-gnu/libicuuc.so.55
libz.so.1 needed by //usr/lib/x86_64-linux-gnu/libxml2.so.2
found libz.so.1 at //lib/x86_64-linux-gnu/libz.so.1
liblzma.so.5 needed by //usr/lib/x86_64-linux-gnu/libxml2.so.2
found liblzma.so.5 at //lib/x86_64-linux-gnu/liblzma.so.5
libm.so.6 needed by //usr/lib/x86_64-linux-gnu/libxml2.so.2
found libm.so.6 at //lib/x86_64-linux-gnu/libm.so.6
ld-linux-x86-64.so.2 needed by //lib/x86_64-linux-gnu/libpthread.so.0
found ld-linux-x86-64.so.2 at //lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
libicudata.so.55 needed by //usr/lib/x86_64-linux-gnu/libicuuc.so.55
found libicudata.so.55 at //usr/lib/x86_64-linux-gnu/libicudata.so.55
libstdc++.so.6 needed by //usr/lib/x86_64-linux-gnu/libicuuc.so.55
found libstdc++.so.6 at //usr/lib/x86_64-linux-gnu/libstdc++.so.6
libgcc_s.so.1 needed by //usr/lib/x86_64-linux-gnu/libicuuc.so.55
found libgcc_s.so.1 at //lib/x86_64-linux-gnu/libgcc_s.so.1
ld: warning: cannot find entry symbol _start; not setting start address

khan2018, 2020-03-09:
Hi again

I am still stuck on the make embedded part. Some help would be appreciated.

The output for make embedded after generating the libiio.a file is 
g++ -O3 -I.. -DVERSION=\"leansdr-1.2.0-27-ge6416aa\" -Wall -Wno-sign-compare -Wno-array-bounds -Wno-unused-variable  -Ilibiio leaniiorx.cc  -lfftw3f -lpthread -Llibiio -L. -liio -lpthread -lxml2 -lz -llzma -static -o leaniiorx.x86_64
/usr/bin/ld: dynamic STT_GNU_IFUNC symbol `strcmp' with pointer equality in `/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libc.a(strcmp.o)' can not be used when making an executable; recompile with -fPIE and relink with -pie
collect2: error: ld returned 1 exit status
Makefile:83: recipe for target 'leaniiorx.x86_64' failed
make: *** [leaniiorx.x86_64] Error 1

pabr, 2020-03-09:
It looks like you are running "make embedded" on a x86_64 machine, so the Makefile is trying to generate a binary for x86_64. If this is really what you want, try the same command without "-static".

But leaniiorx is intended to run on-board the plutosdr. The Makefile can generate a portable (static) plutosdr binary if you run it on another arm-linux board (e.g. raspberry pi with some versions of the OS) with libiio already compiled, but that was a quick hack. The proper approach is documented at https://wiki.analog.com/university/tools/pluto/devs/embedded_code , but that's a lot more work.

You can also test leansdr binaries from leantrx to decide whether you really want to recompile the latest version. Download from https://www.pabr.org/radio/leantrx/leantrx.en.html#plutosdr_net_install

khan2018, 2020-03-12:
Hi

I solved it. I removed the previous installation and did a new clean install and everything worked.
Thanks.

I have another question though, I want to use leandvb to offline decode the dvb-s2 I have pre-recorded. Now the data type I have is from Matlab using AD936x and its in double complex IQ. 
How can I convert it to the data type supported by leandvb.  

I have the following formats I can receive the dvb-s2 in  Matlab:
 1.   int16 — Integer values are the raw 16-bit I and Q samples from the board. The 12-bit value from the ADC of the AD9361 RF chip is sign-extended to 16 bits.

2.    single — Single-precision floating point values are scaled to the range of [–0.0625, 0.0625]. The block derives this value from the sign-extended 16 bits received from the board.

3.    double — Double-precision floating point values are scaled to the range of [–0.0625, 0.0625]. The block derives this value from the sign-extended 16 bits received from the board.

I have been trying to convert using typecast but its not working. 
Thanks !

khan2018, 2020-03-12:
Also, is there a way I can integrate leandvb in my Matlab code? I am looking for a dvb s2 decoder.

pabr, 2020-03-14:
Hopefully a Matlab expert will be able to answer your question. There's probably a way to write IQ data to a file in binary mode using "single" format ("Single-precision floating point values"), then execute leandvb from Matlab with a system() command, then read the resulting MPEG TS file.