pabr/leansdr #25: Which SDR transmitter works with the leandvb receiver?

VDelport, 2020-06-02:
Hello, I am trying to transmit an MPEG-TS with gr-dvbs2 as the transmitter and an rtl-sdr + leandvb as the receiver. I get an almost perfect constellation diagram. The MER is approx. 18 dB. But leandvb does not generate a transport stream. Have you tested leandvb yourself with gr-dvbs2tx? Do leandvb_tx and leandvb_rx work together? That doesn't work for me either.
Best regards.

pabr, 2020-06-03:
Here is the simplest DVB-S2 test you can run:
> make leantsgen leandvbtx leandvb
> ./leantsgen | ./leandvbtx --standard DVB-S2 | ./leandvb --standard DVB-S2 --f32 -f 4e6 --gui -d > /dev/null

You should see a QPSK constellation for signalling, a 8PSK constellation for data, and a steady stream of '_' characters on stderr indicating good frames.

This will rule out compiler/portability issues. Then you can give more details about your use case, command-line parameters, etc.

VDelport, 2020-06-03:
Thank you for your quick reply. The command line fails for me. leandvbtx and leandvb seem to have no option -- standard DVB-S2:

./leantsgen | ./leandvbtx --standard DVB-S2
...
** Error while processing '--standard'

  ./leandvb --help:
...
  --standard S DVB-S (default), DVB-S2 (not implemented)

pabr, 2020-06-03:
Most likely, you need to run `git checkout work` or restart from scratch with `git clone -b work ...` for S2 support.

pabr, 2020-06-03:
It looks like hackrf_transfer expects 8-bit signed ints whereas leandvbtx supports only 16-bit signed ints and 32-bit floats, so it won't work out-of-the-box.

For reception with rtl-sdr there are examples at http://www.pabr.org/radio/qo100sdr .

VDelport, 2020-06-03:
Many thanks for your response. I was now able to successfully test the radio transmission with gr-dvbs2 (github.com/drmpeg/gr-dvbs2) and Hack-RF One as the transmitter and with leandvb and RTL-SDR as the receiver. Thank you for your support and the great project. :) Best regards.

closed 2020-06-06 by VDelport