pabr/leansdr #26: Does lenadvb support CRC for DVB-S2 reception?

VDelport, 2020-07-13:
First of all, sorry for asking this question in the issues.

Broadcast systems use FEC as error protection. However, if it is important that the data is processed without any error after receipt, CRC error protection should also be used, with which bundle errors are identified with a high degree of probability. As far as I know, there is also CRC error protection in DVB-S2. Does lenadvb support CRC for DVB-S2 reception?

pabr, 2020-07-13:
leandvb checks 8-bit CRCs on S2 frame headers but not on individual TS packets. This can be a concern now that people use it near the SNR limit and errors could slip through LDPC+BCH. Let's keep this issue open as a reminder that I need to fix that.

VDelport, 2020-07-13:
Thank you for your quick reply.

VDelport, 2020-07-13:
I have another question that goes in the same direction. If I am correct, there is a flag in the header of a TS packet that is set by a hardware receiver if the TS packet has errors. Does leandvb set the flag too?

pabr, 2020-07-15:
Not currently but the plan is indeed to set the Transport Error Indicator bit on TS packets with a bad CRC.
Implementation is done, will push later this month after testing.

VDelport, 2020-07-16:
>Implementation is done, will push later this month after testing.
That would be great. Many thanks.