pabr/leansdr #28: Some Questions about leansdr in processing IQ file

wufeissdfz, 2020-08-14:
Hello，

Recently，I tried to compile and install leanDVB and finally finish the functional loop tests. 
This week I used leanDVB to process a IQ file sampled from a DVB-S2 standard tx card.
But I failed. I think the IQ file format may cause the problem.
CMD :
` leandvb --s16 -f 2500000 --sr 1000000 --gui --fd-info 2  --standard DVB-S2 --sampler rrc < IQREC-08-12-20-10h17m40s384.iq > test0812_1.ts`
The parameters of the DVB-S2 signal are listed below.
`CCM 1Msps 8PSK 3/5 with pilot ShortFrame `
The payload is a transport stream.
[image]

The IQ file looks well in cooledit, which sample rate is 2.5Msps. (16bits LSB)
[image]

The format of IQ file is complex short. I think the parameter of the IQ file is --s16.
There is the IQ file link. https://we.tl/t-Ak52TAHKwx

Many thanks.

pabr, 2020-08-15:
I can see a strong 8PSK signal in your IQ file but I can't decode it as DVB-S2 either. Have you tested this modulator with another demodulator ?

pabr, 2020-08-15:
The spectrum is inverted in your IQ file. leandvb does not support that yet in DVB-S2 mode.

As a workaround you can flip the sign on each Q sample, or swap each I/Q pair, or remove the first 2 bytes from the file. Then leandvb should produce a 704x576 MPEG2 TS.

pabr, 2020-08-15:
The spectrum is inverted in your IQ file. leandvb does not support that yet in DVB-S2 mode.

As a workaround you can flip the sign on each Q sample or swap each I/Q pair. Then leandvb should produce a 704x576 MPEG2 TS.

wufeissdfz, 2020-08-16:
> The spectrum is inverted in your IQ file. leandvb does not support that yet in DVB-S2 mode.
> 
> As a workaround you can flip the sign on each Q sample or swap each I/Q pair. Then leandvb should produce a 704x576 MPEG2 TS.

Thanks a lot, I will modify the IQ file and try it again.

klimentyev, 2020-09-27:
@wufeissdfz Can you reupload your iq file? I can't download it by your link :( I wanna work with leansdr, but I haven't got any raw iq with DVB-S2.

klimentyev, 2020-09-27:
@pabr I would be grateful if you would share this or a similar IQ file.

pabr, 2020-09-27:
@klimentyev You can generate DVB-S2 IQ streams with tools that ship with leandvb, see:
https://github.com/pabr/leansdr/issues/25#issuecomment-638047435

However this will not simulate the specific issue that was present in the IQ file of @wufeissdfz, i.e. spectrum inversion.

klimentyev, 2020-09-27:
@pabr Thank you very much. The util helps me.

wufeissdfz, 2020-12-05:
> @pabr Thank you very much. The utility helps me.

Sorry for my reply late, I can generate DVB-S2 sample, what about the parameters of DVB-S2 IQ file.

klimentyev, 2021-01-03:
@wufeissdfz Thx, but I wanna process the real signal for check my information data parser.