pabr/leansdr #29: DVB-C implementation

vankxr, 2020-10-21:
Hello, First of all I would like to thank you for releasing this nice piece of software, it has been really helpful in my recent tinkering.
I would like to implement a DVB-S2 to DVB-C transmodulator in software. I have been loking at your code, and at both the DVB standards, and from what I can see, the signal chain is the same up to the outer interleaver block. In DVB-C there is a differential coding on the two MSBs of the symbols, to make the constellation rotation-invariant.
Since you have already implemented the QAM16, QAM64 and QAM256 constellation LUTs, it would only be a matter of implementing the previously mentioned differential coding, am I right? Can you give me some help on this?

Thank you,
João

pabr, 2020-11-02:
The differential coding could be implemented by inserting a block before the constellation transmitter.

But in addition to that, you would need to implement DVB-C-compliant QAM constellations. The QAM constellations in leandvb were added for experimentation only, with naive mappings from bits to QAM symbols.

Also, it looks like DVB-C is obsoleted by DVB-C2, which uses OFDM. Is there really a practical use case for software-defined DVB-C ?

vankxr, 2020-11-02:
DVB-C2 would also work, I just found it was going to be more complicated since OFDM is not used in DVB-S/2