pabr/leansdr #32: Cannot compile ldpc

andimik, 2020-11-12:
Following http://www.pabr.org/radio/leandvb/leandvb.en.html#install building of ldpc fails under Ubuntu 20.04.

```
$ uname -a
Linux E535ubuntu 5.4.0-52-generic #57-Ubuntu SMP Thu Oct 15 10:57:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ LC_ALL=C make CXX=g++ ldpc_tool 
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti tables_handler.cc -c -o tables_handler.o
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti itls_handler.cc -c -o itls_handler.o
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti mods_handler.cc -c -o mods_handler.o
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti ldpc_tool.cc tables_handler.o itls_handler.o mods_handler.o -o ldpc_tool
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::hardN(signed char*, std::complex<float>*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_]+0xe): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_]+0x28): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::mapN(std::complex<float>*, signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4mapNEPS2_Pa]+0x1e): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4mapNEPS2_Pa]+0x61): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::hard(signed char*, std::complex<float>)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_]+0x1a): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_]+0x22): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::map(signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa]+0x25): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa]+0x2d): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 10800>::hardN(signed char*, std::complex<float>*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_]+0xe): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_]+0x28): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 10800>::mapN(std::complex<float>*, signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4mapNEPS2_Pa]+0x1e): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4mapNEPS2_Pa]+0x61): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 10800>::hard(signed char*, std::complex<float>)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4hardEPaS2_]+0x1a): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4hardEPaS2_]+0x22): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 10800>::map(signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE3mapEPa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE3mapEPa]+0x25): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE3mapEPa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE3mapEPa]+0x2d): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 5400>::hardN(signed char*, std::complex<float>*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE5hardNEPaPS2_]+0xe): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE5hardNEPaPS2_]+0x28): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 5400>::mapN(std::complex<float>*, signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4mapNEPS2_Pa]+0x1e): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4mapNEPS2_Pa]+0x61): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 5400>::hard(signed char*, std::complex<float>)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4hardEPaS2_]+0x1a): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi5400EE4hardEPaS2_]+0x22): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
[... cut ...]
/usr/bin/ld: mods_handler.o:mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5softNEPaPS2_f[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5softNEPaPS2_f]+0x64): more undefined references to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw' follow
collect2: error: ld returned 1 exit status
make: *** [Makefile:18: ldpc_tool] Error 1
```

I found a solution at https://github.com/Blockstream/satellite/blob/master/doc/sdr.md but this is not repo.

antonjan, 2021-07-28:
I cant find the solution on the link. what is the solution?

andimik, 2021-07-28:
Honestly, it was long ago and I cannot remember. I guess I had to clone the fork instead.

andimik, 2022-12-01:
I have the same error under Ubuntu 22.04 but I could find a workaround (see below)

```
$ git clone -b ldpc_tool http://github.com/pabr/xdsopl-LDPC-pabr
Klone nach 'xdsopl-LDPC-pabr' …
warning: Umleitung nach https://github.com/pabr/xdsopl-LDPC-pabr/
remote: Enumerating objects: 691, done.
remote: Total 691 (delta 0), reused 0 (delta 0), pack-reused 691
Empfange Objekte: 100% (691/691), 367.23 KiB | 893.00 KiB/s, fertig.
Löse Unterschiede auf: 100% (462/462), fertig.
andreas@E535:~/apps$ cd xdsopl-LDPC-pabr/
andreas@E535:~/apps/xdsopl-LDPC-pabr$ make CXX=g++ ldpc_tool
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti tables_handler.cc -c -o tables_handler.o
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti itls_handler.cc -c -o itls_handler.o
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti mods_handler.cc -c -o mods_handler.o
g++ -std=c++11 -W -Wall -Ofast -fno-exceptions -fno-rtti ldpc_tool.cc tables_handler.o itls_handler.o mods_handler.o -o ldpc_tool
/usr/bin/ld: mods_handler.o: warning: relocation against `_ZN16PhaseShiftKeyingILi8ESt7complexIfEaE6rot_cwE' in read-only section `.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4softEPaS2_f[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4softEPaS2_f]'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::hardN(signed char*, std::complex<float>*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_]+0x24): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE5hardNEPaPS2_]+0x2c): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::mapN(std::complex<float>*, signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4mapNEPS2_Pa]+0x24): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::hard(signed char*, std::complex<float>)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_]+0x1a): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE4hardEPaS2_]+0x22): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 21600>::map(signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa]+0x25): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi21600EE3mapEPa]+0x2d): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 10800>::hardN(signed char*, std::complex<float>*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_]+0x24): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE5hardNEPaPS2_]+0x2c): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 10800>::mapN(std::complex<float>*, signed char*)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4mapNEPS2_Pa[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4mapNEPS2_Pa]+0x24): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_acw'
/usr/bin/ld: mods_handler.o: in function `Modulation<PhaseShiftKeying<8, std::complex<float>, signed char>, 10800>::hard(signed char*, std::complex<float>)':
mods_handler.cc:(.text._ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4hardEPaS2_[_ZN10ModulationI16PhaseShiftKeyingILi8ESt7complexIfEaELi10800EE4hardEPaS2_]+0x1a): undefined reference to `PhaseShiftKeying<8, std::complex<float>, signed char>::rot_cw'
[... cut ...]
-template <typename TYPE, typename CODE>
-constexpr typename TYPE::value_type QuadratureAmplitudeModulation<64, TYPE, CODE>::AMP;
-template <typename TYPE, typename CODE>
-constexpr typename TYPE::value_type QuadratureAmplitudeModulation<256, TYPE, CODE>::AMP;
-template <typename TYPE, typename CODE>
-constexpr typename TYPE::value_type QuadratureAmplitudeModulation<1024, TYPE, CODE>::AMP;
-
 template <int LEN>
 ModulationInterface<complex_type, code_type> *create_modulation(char *name)
 {
```

[image]