pabr/leansdr #34: Some Questions about leansdr in processing DVB-S2 Signal

wufeissdfz, 2020-12-05:
Hello,
Thank you for your help last time.
Recently, I tried to do some test with leandvb by DekTec DTA-2115 DVB-S2 singnal generator.
Sometimes leandvb may stop processing IQ file and exit before read the end of IQ file.
When I used "--ldpc-helper ldpc_tool" to decode, it may cause problem.

./src/apps/leandvb  --inbuf 512000000 --s16 -f 40e6 --sr 20e6 --gui --fd-info 2  --roll-off 0.2 --standard DVB-S2 --sampler rrc --rrc-rej 15 --fastdrift --fastlock --ldpc-helper ldpc_tool --nhelpers 6 -v -d -d < /data/IQREC-08-18-20-10h58m44s85_20M-5s-1f.iq > test.ts

The IQ file link is [https://we.tl/t-05cciHMBFp](https://we.tl/t-05cciHMBFp)

And ldpc_tool output 
`decoder failed at converging to a code word in 25 trials`
[image]

Runtime summary is as follow:
```
.rawiq            : 115M/ 19M      0 writable !, 532512 unread ( 395 3467 395 395 3467 532512 )
.stdin            :  19M/201M 311249598 writable  , 181950423 unread ( 181950423 )
.cnr              :    0/   0      4 writable  ,      0 unread ( 0 )
.spectrum         :    0/   0      4 writable  ,      0 unread ( )
.freq             :    8/   4      3 writable  ,      0 unread ( 0 0 )
.SS               :    8/   4      3 writable  ,      0 unread ( 0 0 )
.MER              :    8/   4      3 writable  ,      0 unread ( 0 0 )
.cstln            : 189k/ 95k 437307 writable  ,    997 unread ( 997 101 )
.PLS cstln        :  96k/ 96k 435942 writable  ,    314 unread ( 314 )
.frame lock       :  158/ 158      2 writable  ,      0 unread ( 0 )
.VBER             :    0/   0      4 writable  ,      0 unread ( )
.lock             :    0/   0      8 writable  ,      0 unread ( 0 )
.locktime         :    0/   0    156 writable  ,      0 unread ( 0 )
.TS packets       :    0/   0    156 writable  ,      0 unread ( 0 0 )
.Bits processed   :    0/   0    156 writable  ,      0 unread ( 0 )
.Bits corrected   :    0/   0    156 writable  ,      0 unread ( 0 )
.VBER             :    0/   0      4 writable  ,      0 unread ( 0 )
.packet counter   :    0/   0    156 writable  ,      0 unread ( 0 )
.PL slots         :  94k/ 95k    336 writable !,   1104 unread ( 1104 )
.BB frames        :    0/   0      4 writable  ,      0 unread ( 0 )
.FEC frames       : 1024/1028      0 writable !,      4 unread ( 4 )
Total buffer memory: 2015908 KiB
```

When I stopped to decode, it can be run to end normally.

Runtime summary is as follow:
```
 .rawiq            : 1207M/201M   7375 writable  , 118463 unread ( 354 1378 354 354 1378 118463 )
.stdin            : 201M/201M 311249598 writable  ,      0 unread ( 0 )
.cnr              :    0/   0      4 writable  ,      0 unread ( 0 )
.spectrum         :    0/   5      3 writable  ,      0 unread ( )
.freq             :   92/  46      3 writable  ,      0 unread ( 0 0 )
.SS               :   92/  46      3 writable  ,      0 unread ( 0 0 )
.MER              :   92/  46      3 writable  ,      0 unread ( 0 0 )
.cstln            :   2M/  1M  43928 writable  ,    136 unread ( 136 8 )
.PLS cstln        : 645k/646k 416924 writable  ,    900 unread ( 900 )
.frame lock       : 1906/1906      1 writable  ,      0 unread ( 0 )
.VBER             :    0/   0      4 writable  ,      0 unread ( )
.lock             :    2/   2      6 writable  ,      0 unread ( 0 )
.locktime         :  10k/ 10k     87 writable  ,      0 unread ( 0 )
.TS packets       :  21k/ 10k     87 writable  ,      0 unread ( 0 0 )
.Bits processed   : 6939/6939     34 writable  ,      0 unread ( 0 )
.Bits corrected   : 6939/6939     34 writable  ,      0 unread ( 0 )
.VBER             : 1509/1509      3 writable  ,      0 unread ( 0 )
.packet counter   :  343/ 343    125 writable  ,      0 unread ( 0 )
.PL slots         :   1M/  1M    738 writable  ,      0 unread ( 0 )
.BB frames        : 1367/1367      3 writable  ,      0 unread ( 0 )
.FEC frames       : 6939/6939      2 writable  ,      0 unread ( 0 )
Total buffer memory: 2015686 KiB
```
Please help me to look into this together.  Thanks a lot.

pabr, 2020-12-05:
leandvb with ldpc_tool has known issues when the signal is very noisy (or when there is pure noise before/after a strong signal, as in your recording).

As a workaround you can try adding "--modcods 0x80000 --framesizes 2" to the command-line. You should get at least 1.5 MB of MPEG TS from your recording. Please let me know if this helps.

wufeissdfz, 2020-12-06:
> leandvb with ldpc_tool has known issues when the signal is very noisy (or when there is pure noise before/after a strong signal, as in your recording).
>
> As a workaround you can try adding "--modcods [...]

Thanks a lot, I am trying to fix it as you mentioned.

wufeissdfz, 2020-12-16:
> leandvb with ldpc_tool has known issues when the signal is very noisy (or when there is pure noise before/after a strong signal, as in your recording).
>
> As a workaround you can try adding "--modcods [...]

After I used these parameters to filter mocods and framesizes sometimes it works well.

wufeissdfz, 2020-12-16:
Thank you for your last reply.

Recently, I used leandvb to process some DVB-S2 ACM sample, sometimes it works well.
But sometimes the signal C/N is almost better than 15.0dB，when I used this workaround to process some sample,  it worked with ldpc_tool may stop processing IQ file and exit before read the end of IQ file.

The IQ file link is https://we.tl/t-Q82VKIzHYx

wufeissdfz, 2020-12-16:
Thank you for your last reply.

Recently, I used leandvb to process some DVB-S2 ACM sample, sometimes it works well.
But sometimes the signal C/N is almost better than 15.0dB，when I used this workaround to process some sample, it worked with ldpc_tool may stop processing IQ file and exit before read the end of IQ file.

The IQ file link is https://we.tl/t-Q82VKIzHYx.

wufeissdfz, 2020-12-16:
Thank you for your last reply.

Recently, I tried to process some DVB-S2 ACM sample with leandvb, sometimes it works well.

But somtimes the signal C/N is almost better than 15.0dB, I used the workaround which we used to filter modcods and framesizes, leandvb with ldpc_tools may stop and exit before EOF.

`"--inbuf 32000000 --s16 -f 50e6 --sr 29.09e6 --gui --fd-info 2 --roll-off 0.2 --standard DVB-S2 --sampler rrc  --rrc-rej 30 -v -d -d --fastdrift --framesizes 2 --fd-gse 3 --ldpc-helper ldpc_tool --buf-factor 8"`

The IQ file link is https://we.tl/t-Q82VKIzHYx.

Please help me to look into this together. Thanks a lot.

pabr, 2020-12-16:
If you see this on stderr, you must adjust /proc/sys/fs/pipe-max-size. Besides, you may want to multiply the value by the number of ACM modcods.
> *** Failed to increase pipe size.
> *** Try echo 2073600 > /proc/sys/fs/pipe-max-size
> *** Throughput will be suboptimal.

You also need to apply the --modcods workaround to select one modcod. With "--modcods 0x40000" I get 350 kB of binary data from your recording. With "--modcods 0x1000", I get 75 kB.

Do you have a use case that requires decoding multiple modcods simultaneously ? I am afraid the current version cannot do that reliably.