pabr/leansdr #35: Issues transmitting DVB-S2 with leandvbtx

vankxr, 2021-01-19:
Hello,
I have been trying to transmit DVB-S2 using leandvbtx, but I am running into some problems. I have narrowed it down to leandvbtx's DVB-S2 mode itself by doing some tests:

- DATV-Express DVB-S2 TX -> SDR Angel DVB-S2 RX - OK
- DATV-Express DVB-S2 TX -> leandvb DVB-S2 RX - OK
- DATV-Express DVB-S TX -> SDR Angel DVB-S RX - OK
- DATV-Express DVB-S TX -> leandvb DVB-S RX - OK
- leandvbtx DVB-S TX -> SDR Angel DVB-S RX - OK
- leandvbtx DVB-S TX -> leandvb DVB-S RX - OK
- leandvbtx DVB-S2 TX -> SDR Angel DVB-S2 RX - NOT WORKING
- leandvbtx DVB-S2 TX -> leandvb DVB-S2 RX - OK (if piped directly, i.e., software loopback, even with awgn noise added), NOT WORKING (if transmitted to SDR and received again, i.e., analog RF loopback)

I have increased my pipe size, before I did it there were lost samples somewhere, but now there are none, the transmission is continuous.

Did you ever test leandvbtx DVB-S2 mode extensively? I can see an issue where it is reported that the tests do not run correctly, but even then, according to my results, software loopback works, even with white noise added. It stops working once I loop the signal back in the analog domain.

Thank you.

pabr, 2021-01-20:
The open issue about self-tests is a trade-off between low SNR and higher-order MODCODS. It's hard to handle all cases equally well. Probably unrelated unless you are trying to do 32APSK.

Another test you can do is leandvbtx->file1 then file1->SDR->RF->SDR->file2 then file2->leandvbrx.  This would rule out any realtime performance issue.

Can you show your command lines ? Maybe post a IQ recording of the RX side ?

vankxr, 2021-01-20:
Hello,
I tried to pipe the TX IQ into a file, then transmit the file, and also to pipe the RX IQ to a file, then run the demodulator on the file data, with no success.
The command line looks like this:
`sudo leandvbtx \
    --s16 \
    -f 3 \
    --standard DVB-S2 \
    --modcod 14 \
    --roll-off 0.20 \
    --rrc-rej 40 \
    --buf-factor 2
    
sudo leandvb \
    --u8 \
    -f 960e3 \
    --sr 256e3 \
    --standard DVB-S2 \
    --sampler linear \
    --ldpc-helper ldpc_tool \
    --nhelpers 6 \
    --buf-factor 4 \
    --gui`

Here is an IQ file directly from RTL-SDR. unsigned 8 bit data, 960 kHz sample rate, 256 kSymbols/s 8PSK 3/4 DVB-S2 centered at 0Hz: https://easyupload.io/9mqtir
The resulting Transport Stream should have a bitrate of 570 kbps and should contain only one channel with a static image 640x480 pixels.[https://easyupload.io/9mqtir](https://easyupload.io/9mqtir)

pabr, 2021-01-22:
Works for me with "--sr 256060" instead of "--sr 256e3". This suggests that the oscillator frequency error from your SDRs (rx+tx) is about 230 ppm, which is a lot (the default tolerance is 100 ppm). Solutions:

- Specify --sr as above to compensate the oscillator error
- Add "--sr-tol 300e-6" (undocumented) to increase the tolerance to 300 ppm
- Calibrate the oscillator correction factor (hardware-specific).

Besides, I recommend "--sampler rrc" for noisy channels.

vankxr, 2021-01-23:
Hello,

Indeed, my SDR had a bug in the software that controls the board clocks, and the frequency was slightly off.
I dug the code and fixed the problem, and now everything works as expected, even in real-time, without changing the tolerance.

Thank you for you help !

closed 2021-01-23 by vankxr