pabr/leansdr #36: Leandvb integration

i1ndp, 2021-12-09:
I am trying to build a DTV receiver using a B200 Ettus receiver , the main problem to me was to find a digital decoder but finally dropped into leandvb. I did some test using the decoder as an application and apparently it was working but was not exactly what i was looking for.
I decided then to use the source as add on to mine adding the source files to my application changing very few to start it as a thread and using an unnamed pipe to transfer data.Problems?  All easy until  start testing.
leandvb returns from the scheduler after having done very few steps.
The following is a trace of the activity (leandvb runnables) when i start to send data:

Runnable: 0 stdin
Runnable: 1 scaler
Runnable: 2 spectrum
Runnable: 3 TS packets
Runnable: 4 rate_estimator
Runnable: 5 S2 frame receiver
Runnable: 6 S2 deinterleaver
Runnable: 7 S2 fecdec io
Runnable: 8 S2 deframer
sch stopped---------------------------stop and start closing
Runnable x shutdown: 0 stdin
Runnable x shutdown: 1 scaler
Runnable x shutdown: 2 spectrum
Runnable x shutdown: 3 TS packets
Runnable x shutdown: 4 rate_estimator
Runnable x shutdown: 5 S2 frame receiver
Runnable x shutdown: 6 S2 deinterleaver
Runnable x shutdown: 7 S2 fecdec io
Runnable x shutdown: 8 S2 deframer

The code is very hermetic and i was not able to go any further ..
As help the following is the used configuration:

config()
    : verbose(false),
      debug(true),
      debug2(true),
      highspeed(false),
      input_format(INPUT_F32),
      float_scale(1.0),
      loop_input(false),
      input_pipe(0),
      input_buffer(0),
      buf_factor(4),
      Fs(2.4e6),
      Fderot(0),
      anf(0),
      cnr(false),
      decim(0),
      fd_pp(-1),
      fd_gse(-1),
      fd_iqsymbols(-1),
      awgn(0),
      Fm(2e6),
      standard(DVB_S2),
      constellation(cstln_base::QPSK),
      strongpls(false),
      modcods(0xffffffff),
      framesizes(0x03),
      fec(FEC12),
      Ftune(0),
      allow_drift(false),
      freq_tol(0.25),
      sr_tol(100e-6),
      fastlock(true),
      fastdrift(false),
      viterbi(true),
      hard_metric(false),
      ldpc_bf(10),
      ldpc_helper("ldpc_tool"),
      nhelpers(4),
      resample(false),
      resample_rej(10),
      sampler(config::SAMP_RRC),
      rrc_steps(0),
      rrc_rej(10),
      rolloff(0.35),

      hdlc(false),
      packetized(false),

      gui(false),
      duration(60),
      linger(false),
      fd_info(-1),
      Finfo(5),
      fd_const(-1),
      fd_spectrum(-1),
      json(false) {

Any hints will be very appreciated.
Thanks in advance,nando

pabr, 2021-12-09:
Your approach of feeding data to a leandvb thread through a pipe should work.

The scheduler stops when the flowgraph is not making any progress. Normally this means EOF on stdin or stdout. Are you redirecting stdout to a pipe as well ?

Can you copy-paste the debug traces from stderr, including the final statistics report ? Please configure verbose(true).

There are known issues with the external LDPC decoder when signals are noisy. You may want to try with ldpc_helper(NULL) first.

i1ndp, 2021-12-09:
Thanks for the answer , attached there is a file done with verbose and 
debug 2 enabled.
I tried already what to me were the significant configuration 
alternative including to remove the helper but the application does not 
reach the code instantiating the helper. The output is supposed to be 
the vlc with the receiving side of a pipe while the writer side is moved 
in the standard output of the vlc father but, to be frank, i was not 
able to identify the code  dealing with the output.
I have to add that the test was not done with significant data (sdr not 
connected to the antenna) but again the behavior would be the same.
nando

[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-0-unknown
[INFO] [B200] Detected Device: B200mini
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 38.400000 MHz... 
[INFO] [B200] Actually got clock rate 38.400000 MHz.
Wait for REF... - 1 REF Locked: ...
[INFO] [B200] Asking for clock rate 38.400000 MHz... 
[INFO] [B200] OK
[INFO] [B200] Asking for clock rate 38.400000 MHz... 
[INFO] [B200] OK
Measuring SPD
Roll-off 0.35
Fs after resampling/decimation: 2400000.000000 Hz
RRC interpolator: 53 steps
RRC filter: 167 coeffs.
rrc = [ -0.149261, -0.150110, -0.150489, -0.150378, -0.149760, -0.148621, -0.146944, -0.144715, -0.141922, -0.138552, -0.134595, -0.130040, -0.124881, -0.119108, -0.112718, -0.105705, -0.098066, -0.089800, -0.080907, -0.071388, -0.061245, -0.050484, -0.039110, -0.027130, -0.014553, -0.001390, 0.012348, 0.026646, 0.041490, 0.056864, 0.072751, 0.089129, 0.105979, 0.123280, 0.141007, 0.159136, 0.177641, 0.196496, 0.215669, 0.235136, 0.254865, 0.274823, 0.294980, 0.315301, 0.335755, 0.356306, 0.376919, 0.397559, 0.418190, 0.438777, 0.459281, 0.479668, 0.499899, 0.519939, 0.539749, 0.559295, 0.578540, 0.597447, 0.615981, 0.634107, 0.651790, 0.668997, 0.685694, 0.701850, 0.717433, 0.732413, 0.746760, 0.760446, 0.773445, 0.785731, 0.797279, 0.808067, 0.818073, 0.827278, 0.835662, 0.843210, 0.849906, 0.855736, 0.860690, 0.864756, 0.867928, 0.870197, 0.871561, 0.872016, 0.871561, 0.870197, 0.867928, 0.864756, 0.860690, 0.855736, 0.849906, 0.843210, 0.835662, 0.827278, 0.818073, 0.808067, 0.797279, 0.785731, 0.773445, 0.760446, 0.746760, 0.732413, 0.717433, 0.701850, 0.685694, 0.668997, 0.651790, 0.634107, 0.615981, 0.597447, 0.578540, 0.559295, 0.539749, 0.519939, 0.499899, 0.479668, 0.459281, 0.438777, 0.418190, 0.397559, 0.376919, 0.356306, 0.335755, 0.315301, 0.294980, 0.274823, 0.254865, 0.235136, 0.215669, 0.196496, 0.177641, 0.159136, 0.141007, 0.123280, 0.105979, 0.089129, 0.072751, 0.056864, 0.041490, 0.026646, 0.012348, -0.001390, -0.014553, -0.027130, -0.039110, -0.050484, -0.061245, -0.071388, -0.080907, -0.089800, -0.098066, -0.105705, -0.112718, -0.119108, -0.124881, -0.130040, -0.134595, -0.138552, -0.141922, -0.144715, -0.146944, -0.148621, -0.149760, -0.150378, -0.150489, -0.150110, -0.149261 ];
Verso FEC-protected frames nhelpers 4  
Output:
  '_': packet received without errdeframerors
  '.': error-corrected packet
  '!': packet with remaining errors
Runnable: 0 stdin
Runnable: 1 scaler
Runnable: 2 spectrum
Runnable: 3 TS packets
Runnable: 4 rate_estimator
Runnable: 5 S2 frame receiver
Runnable: 6 S2 deinterleaver
Runnable: 7 S2 fecdec io
Runnable: 8 S2 deframer
sch stopped
Runnable x shutdown: 0 stdin
Runnable x shutdown: 1 scaler
Runnable x shutdown: 2 spectrum
Runnable x shutdown: 3 TS packets
Runnable x shutdown: 4 rate_estimator
Runnable x shutdown: 5 S2 frame receiver
PL errors: 0/0 (-nan ppm)
Runnable x shutdown: 6 S2 deinterleaver
Runnable x shutdown: 7 S2 fecdec io
Runnable x shutdown: 8 S2 deframer

.rawiq            :    0/   0 319507 writable  ,      0 unread ( 0 0 )
.stdin            :    0/   0 319507 writable  ,      0 unread ( 0 )
.cnr              :    0/   0      4 writable  ,      0 unread ( )
.spectrum         :    0/   0      4 writable  ,      0 unread ( )
.freq             :    0/   0      4 writable  ,      0 unread ( )
.SS               :    0/   0      4 writable  ,      0 unread ( )
.MER              :    0/   0      4 writable  ,      0 unread ( )
.cstln            :    0/   0 319507 writable  ,      0 unread ( )
.PLS cstln        :    0/   0 319507 writable  ,      0 unread ( )
.frame lock       :    0/   0      4 writable  ,      0 unread ( )
.VBER             :    0/   0      4 writable  ,      0 unread ( )
.lock             :    0/   0      8 writable  ,      0 unread ( )
.locktime         :    0/   0    156 writable  ,      0 unread ( )
.TS packets       :    0/   0    156 writable  ,      0 unread ( 0 )
.Bits processed   :    0/   0    156 writable  ,      0 unread ( 0 )
.Bits corrected   :    0/   0    156 writable  ,      0 unread ( 0 )
.VBER             :    0/   0      4 writable  ,      0 unread ( )
.PL slots         :    0/   0   1440 writable  ,      0 unread ( 0 )
.BB frames        :    0/   0      4 writable  ,      0 unread ( 0 )
.FEC frames       :    0/   0      4 writable  ,      0 unread ( 0 )

pabr, 2021-12-09:
According to your stderr log, leandvb is not reading any IQ data. You could run your program under "strace -f" to see if it is doing the I/O syscalls you expect, and if the file descriptors for the pipe are correct.

i1ndp, 2021-12-10:
Yes, i forgot to tell before that the error i get is broken pipe for 
each i/q block write. What i though is that was the effect (not the 
cause) of the leandvb termination.
For sure there  is a big error on my side but enable to understand why.
I am supposed to use an unnamed pipe created by the father in a common 
area, initially i tried no to touch the stdin file modifying the 
following leandvb code:

new file_reader<cf32>(sch,*leanarea.fdlean[0]*, *p_stdin); I collected 
the trace file as you suggested

while the original was pointing to stdin:

new file_reader<cf32>(sch,0, *p_stdin);

Then, suspecting it was not the only file reader  i copied the pipe 
input file descriptor to the 0 file descriptor and restoring the 
original code , same result !
I collected the trace file as you suggested but is refused by the email 
being too big (57MB) in any case i doubt it could be of some helpbeing 
saturated by usb activity and received band monitor activity.
Not easy for me to insulate  the leandvb interface.
What i want to try now is to avoid starting leandvb and generate a dummy 
thread instead to receive the I/Q blocks