pabr/leansdr #37: Leandvb output UDP

f4dvk, 2022-09-23:
Hello,

Is it possible to direct the TS signal to a UDP address?
Example of UDP operation:
https://github.com/BritishAmateurTelevisionClub/longmynd

Thank you,
Stephane.

pabr, 2022-09-27:
No, UDP output is not currently implemented. Some users pipe leandvb output into another command-line tool (tsduck ?) to split the TS into 188-byte chunks and send them over UDP.

pabr, 2022-09-28:
Oh well, it's implemented now. Try "leandvb --ts-udp IPADDR:PORT". Tested with "mplayer udp://127.0.0.1:4000".

F5OEO, 2022-09-28:
Nice, but warn : TS over UDP is generally standardized with a fixed payload of 7*TSpackets to fit in an ethernet frame.

pabr, 2022-09-28:
Thanks for the tip, should be fixed now.

f4dvk, 2022-09-28:
Thanks for the implementation. :-).
Stephane.

closed 2023-10-25 by f4dvk