pabr/leansdr #39: Processing IQ file to completion with nhelpers > 1

igorauad, 2023-03-09:
Hi @pabr ,

I'm trying to benchmark the [gr-dvbs2rx](https://github.com/igorauad/gr-dvbs2rx) performance and compare it to leandvb based on your `work` branch. For example, I'm using the command below for one of the IQ files I have:

```
leandvb --standard DVB-S2 --nhelpers 6 --ldpc-helper /usr/local/bin/ldpc_tool --sampler rrc --rrc-rej 30 --u8 --modcods 32 -f 2000000 --sr 1000000 --roll-off 0.2 --framesizes 1 < qpsk35.iq > /dev/null -d -v
```

The command works OK and demodulates the DVB-S2 signal. The problem is that `leandvb` hangs in the end instead of exiting once the input IQ file reaches EOF. I have realized the problem only occurs with `--nhelpers` greater than one, so I suspect it could be hanging in [this loop](https://github.com/pabr/leansdr/blob/work/src/leansdr/dvbs2.h#L2231).

Any hints? Is there a way to prevent this via command-line options? I would still like to run with more than one helper to put leandvb in a fast configuration for a fair comparison with gr-dvbs2rx.

Regards

pabr, 2023-03-10:
On startup, does leandvb print warnings to stderr about tweaking /proc/sys/fs/pipe-max-size and/or /proc/sys/fs/pipe-user-pages-soft ? Insufficient buffering could explain deadlocks.

Feel free to share qpsk35.iq if you'd like me to test.

pabr, 2023-03-10:
For this to work you need to also patch ldpc_tool so as to reduce its latency from 32 to 16 frames. Maybe changing BLOCKS from 32 to 16 will work, but that might sacrifice half the SIMD benefits.
https://github.com/pabr/xdsopl-LDPC-pabr/blob/ldpc_tool/ldpc_tool.cc#L105

igorauad, 2023-03-10:
Thanks a lot, @pabr . That works. I understand the SIMD performance will be worse, but it may be a fair comparison given that the gr-dvbs2rx LDPC decoder also processes 16 blocks at a time in the ARM arch I'm using for testing (configured here: https://github.com/igorauad/gr-dvbs2rx/blob/master/lib/ldpc_decoder_cb_impl.cc#L652).

I appreciate the help. 

Regards

closed 2023-03-10 by igorauad